seandepagnier/weatherfax_pi #12: Build error on EL6 x86_64

SethDart, 2015-01-23:
I'm packaging your plugin for Fedora & EPEL. All works but EL6 x86_64. Here's the error:
`/builddir/build/BUILD/opencpn-plugin-weatherfax-1.3/src/WeatherFaxUI.cpp: In constructor 'WeatherFaxBase::WeatherFaxBase(wxWindow*, wxWindowID, const wxString&, const wxPoint&, const wxSize&, long int)':
/builddir/build/BUILD/opencpn-plugin-weatherfax-1.3/src/WeatherFaxUI.cpp:130: error: invalid static_cast from type '<unresolved overloaded function type>' to type 'void (wxEvtHandler::*)(wxCloseEvent&)'
/builddir/build/BUILD/opencpn-plugin-weatherfax-1.3/src/WeatherFaxUI.cpp:158: error: invalid static_cast from type '<unresolved overloaded function type>' to type 'void (wxEvtHandler::*)(wxCommandEvent&)'
/builddir/build/BUILD/opencpn-plugin-weatherfax-1.3/src/WeatherFaxUI.cpp: In destructor 'virtual WeatherFaxBase::~WeatherFaxBase()':
/builddir/build/BUILD/opencpn-plugin-weatherfax-1.3/src/WeatherFaxUI.cpp:168: error: invalid static_cast from type '<unresolved overloaded function type>' to type 'void (wxEvtHandler::*)(wxCloseEvent&)'
/builddir/build/BUILD/opencpn-plugin-weatherfax-1.3/src/WeatherFaxUI.cpp:196: error: invalid static_cast from type '<unresolved overloaded function type>' to type 'void (wxEvtHandler::*)(wxCommandEvent&)'
/builddir/build/BUILD/opencpn-plugin-weatherfax-1.3/src/WeatherFaxUI.cpp: In constructor 'InternetRetrievalDialogBase::InternetRetrievalDialogBase(wxWindow*, wxWindowID, const wxString&, const wxPoint&, const wxSize&, long int)':
/builddir/build/BUILD/opencpn-plugin-weatherfax-1.3/src/WeatherFaxUI.cpp:713: error: invalid static_cast from type '<unresolved overloaded function type>' to type 'void (wxEvtHandler::*)(wxMouseEvent&)'
/builddir/build/BUILD/opencpn-plugin-weatherfax-1.3/src/WeatherFaxUI.cpp:722: error: invalid static_cast from type '<unresolved overloaded function type>' to type 'void (wxEvtHandler::*)(wxCommandEvent&)'
/builddir/build/BUILD/opencpn-plugin-weatherfax-1.3/src/WeatherFaxUI.cpp:723: error: invalid static_cast from type '<unresolved overloaded function type>' to type 'void (wxEvtHandler::*)(wxCommandEvent&)'
/builddir/build/BUILD/opencpn-plugin-weatherfax-1.3/src/WeatherFaxUI.cpp: In destructor 'virtual InternetRetrievalDialogBase::~InternetRetrievalDialogBase()':
/builddir/build/BUILD/opencpn-plugin-weatherfax-1.3/src/WeatherFaxUI.cpp:735: error: invalid static_cast from type '<unresolved overloaded function type>' to type 'void (wxEvtHandler::*)(wxMouseEvent&)'
/builddir/build/BUILD/opencpn-plugin-weatherfax-1.3/src/WeatherFaxUI.cpp:744: error: invalid static_cast from type '<unresolved overloaded function type>' to type 'void (wxEvtHandler::*)(wxCommandEvent&)'
/builddir/build/BUILD/opencpn-plugin-weatherfax-1.3/src/WeatherFaxUI.cpp:745: error: invalid static_cast from type '<unresolved overloaded function type>' to type 'void (wxEvtHandler::*)(wxCommandEvent&)'`

The first four errors are easy:
`- this->Connect( wxEVT_CLOSE_WINDOW, wxCloseEventHandler( WeatherFaxBase::OnClose ) );
+ this->Connect( wxEVT_CLOSE_WINDOW, wxCloseEventHandler( WeatherFaxBase::OnClose ), NULL, this );`

But I've not found the others. As this code is generated with wxFormBuilder I don't know what to do.

I'll not be able to package this plugin for EL6 for now :-(

rgleason, 2015-01-24:
Sethdart, just a stupid question here. Do you need the supplemental files?  Look at the Readme, it might be related to having to download from opencpnplugins some libs.

seandepagnier, 2015-01-24:
so...  wxformbuilder is broken or gcc?

What version of gcc?  Does it work on 32bit? I am using 64bit linux:
gcc version 4.8.1 (Ubuntu/Linaro 4.8.1-10ubuntu9)

Maybe it's crazy to consider, but did you try llvm?  I used that compiler
with perfect success drop in replacement for gcc as well.

Sean

SethDart, 2015-01-24:
rgleason, the libs are required for Windows only. Definitely not the issue here ;-)

I'm using the mock toolchain used by all Fedora/EPEL packagers. I've only created a script to handle all the packaging, rpm signing and repo update.
It works like a charm for all Fedora build (32bits & 64bits) as well as EPEL 7 (64bits only)

The only fail is for EPEL 6 (64bits only. 32bits untested)

The mock toolchain creates a chroots then and builds packages in them with the exact package versioning this distro has. So the issue should arrise on baremetal EPEL 6 x86_64

Here's the package version for EPEL 6 x86_64
 cmake28-2.8.11.2-1.el6.x86_64
 gcc-c++-4.4.7-11.el6.x86_64
 wxGTK-devel-2.8.12-1.el6.x86_64

I see that gcc is way outdated, probably the curlpit here.

SethDart, 2015-01-24:
I just complete a test on a CentOS 6.5 VM... same result :-(

I'll try to find a repo with more recent gcc and report

rgleason, 2018-01-19:
This is pretty old now. 2015 ---> 2018.  Has it been resolved or not?  We have new versions....

nohal, 2018-01-19:
And there was no change in the toolchain on EL6. What did change is we do not support wx2.8 anymore so we effectively do not support EL6.

closed 2018-02-08 by seandepagnier