seandepagnier/weatherfax_pi #16: User information

ptulp, 2015-05-07:
Sean,

How can we inform the user that for OSX the portaudio libraries should be installed.
Otherwise the plugin would not show up and cause unnecessary problems.
For OSX this would be by the  "brew install portaudio" command in a terminal window.
Should we mention this on the download page of the plugin??

Peter

nohal, 2015-05-07:
Peter...
We should find a way to install the library with the plugin - it is unrealistic to expect the users being able to install anything manually. As we bitterly realized over the time...

Pavel

ptulp, 2015-05-08:
Pavel,
The problem is that I am not sure there one libportaudio that fits all OSX versions.
I cannot find a download of a compiled library for OSX. All procedures I find do a download of the source and do a compile/install.
We could do a check if the library is present before installing the package with a procedure of how to
install the library and stop the install if not present.
That would at least prevent the plugin not showing up in openCPN as it would not load the plugin.

Peter

seandepagnier, 2015-05-08:
Are you serious that one libportaudio can't fit all versions?  Why not?  We could make the plugin dlopen portaudio and then if that fails it can still work without capture support.

ptulp, 2015-05-08:
This is just an assumption as I cannot find a precompiled library anywhere.
As I don't have multiple OSX machines I cannot test this.

I have made a small change to the buildosx/InstallOSX/weatherfax_pi.pkgproj.in file which will display
the following message if the library is not found:

Please install PortAudio. 
Install Homebrew, following the instructions on http://brew.sh. 
Install portaudio with command: brew install portaudio in a terminal window

It then aborts the installation.
If we use your dlopen solution then probably all OSX users will miss the capture function and there will be questions about it so then we need to mention this somewhere.

seandepagnier, 2015-05-08:
Could you instead just take the library distributed with homebrew, and include it in the package?

nohal, 2015-05-08:
As far as I can tell we should be good just packaging the dylib (which has to be built targeting 10.7 as all the other stuff we distribute)

ptulp, 2015-05-08:
OK, after more searching I found a 10.7 portaudio library.
It seems to work if I replace my original libraries with it.
So now I need to include it in the OSX install file.
Where should we place the lib file. Create a directory libs or images?

nohal, 2015-05-08:
I suppose there is a standard location (where the wx dylibs are placed by the OpenCPN package) or maybe next to the _pi dylib. I know close to nothing about how the OS X app bundles work though.

ptulp, 2015-05-08:
I was more wondering where to put the source not the destination.
Can we put this source library here in github or is that not the place?

nohal, 2015-05-08:
I thought you talk about a dylib already built for 10.7... If it is source, different from upstream and "hard" to find, just create a github repo for it.
If it really is a prebuilt dylib, place it on http://sourceforge.net/projects/opencpnplugins/ and we will mention it in the README (we use the same procedure for the libraries needed on windows already). During packaging, it should be in the build directory (or maybe in buildosx, we just have to reflect it in the packaging instructions).

ptulp, 2015-05-08:
OK, so because this is for weatherfax I should put it in http://sourceforge.net/projects/opencpnplugins/files/weatherfax_pi/   Right?

nohal, 2015-05-08:
Yes, please.

lanceberc, 2015-05-08:
I used ports to install portaudio on OSX Yosemite ("sudo port install portaudio") - this might be easier than brew for some people. Document both?

How hard would it be to disable this feature if portaudio isn't present? Do you think a significant fraction of OpenCPN users will be able to hook a shortwave/SSB radio to their computer?

nohal, 2015-05-08:
The majority of normal users is not able to use either. As we can package the dylib for them, there should be no problem.

ptulp, 2015-05-08:
I made an update to include the library in the package.
Also put a zip file with the header and library files on sourceforge for download before a compile and packaging

lanceberc, 2015-05-11:
I was able to use the sourceforge package to build/install the plugin on OSX-10.10.

I have notes for my install experience, but the forum system has decided that I'm suspicious and won't let me post. A tad frustrating.

rgleason, 2016-06-15:
lanceberc, can you send these notes to me, I will attach.  You have my email I believe (PM)

lanceberc, 2016-06-15:
Have lost your PM, but asked for it in message on CF.

rgleason, 2017-02-18:
lanceberc, have you perhaps seen these notes for Mac install of wxfax_pi?
I can include them somehow in documentation?

rgleason, 2017-08-12:
Can this be closed now? Is better documentation really needed? I believe what is needed should be in the  readme.

lanceberc, 2017-08-13:
I'd updated the Mac build notes a few summers ago but I don't think anyone has maintained them since.

rgleason, 2017-09-14:
Lanceberc,
Would you like to attach them to here, we can include them in the README.

rgleason, 2018-01-19:
Peter,
Can this be closed now?  I think you guys resolved it.  Thanks.

nohal, 2018-01-19:
Did we? How? If you refer to my latest PR, it did nothing about this issue.

ptulp, 2018-01-19:
Rick,

Pavel is right, there are still issues with it.

Peter

nohal, 2018-01-19:
I hope https://github.com/seandepagnier/weatherfax_pi/pull/85/commits/2fc36396591fd8583abbf9aa90cb0e1afd67f3aa solves this.

rgleason, 2018-01-29:
Have any Mac Users had a chance to test Pavel's improvements? If so, can this be closed now?  Sean has accepted the PR.

closed 2018-02-08 by seandepagnier