seandepagnier/weatherfax_pi #24: please tag releases

mschiff, 2016-03-29:
Dear developer,

I would like to make your plugin available to the users of Gentoo Linux.

Right now this is not possible because there are no releases available for your plugin.

It would be very usefull to have tagged releases, so the source for a specific version of the plugin can be downloaded. Please add a release/tag for your released versions. This makes packaging a lot easier.

Please see: [Creating Releases](https://help.github.com/articles/creating-releases/)

Thanks in advance!

mschiff, 2016-12-30:
Ping?

rgleason, 2017-09-15:
git tag  ---> shows v1.3
I think the plugin is at v1.7.003 assuming Sean accepts some PR.

It would not be difficult to 

 git tag -a v1.7.003 -m "Xml update, fix menu + preferences bug"
against Sean's Master branch after he merges the 3 PR, but would that work?
Then make a Pull Request, but shouldn't this be done by the Dev?

Rgleason

rgleason, 2018-01-19:
Can this be closed now?  I can't close it. I think Sean can.

mschiff, 2018-02-02:
Thanks @rgleason for your info, but there are still no releases here :-/

You have a version 1.7007 (not 1.7.007!) in your fork, but this does not compile for me on linux.

Generally its best to have defined releases from the author or the new maintainer of a software. So why are there two forks now? Who is the current official maintainer of this plugin now?

TIA

-Marc

rgleason, 2018-02-02:
Marc there are not two forks. I compile for windows only.

nohal, 2018-02-02:
It would probably be enough to turn on the Appveyor integration contributed 3 months ago and not do continuous integration builds manually. It is a pure waste of time and effort...

rgleason, 2018-02-02:
I agree. Isn't this something Sean does? I have not accepted the appveyor offer yet, but could. No good deed goes unpunished around here.

nohal, 2018-02-02:
Yes, it is something Sean should do. Like tagging the releases. :)

seandepagnier, 2018-02-06:
> It would probably be enough to turn on the Appveyor integration contributed 3 months ago

What appveyor integration was contributed 3 months ago?  To this repository, or to some other opencpn plugin?

nohal, 2018-02-06:
To this repository, you merged it in https://github.com/seandepagnier/weatherfax_pi/pull/73
But to make it actually do something you also have to do the other half - create the accounts on https://www.appveyor.com and https://travis-ci.org, integrate them with your Github account and enable build for this repository.

rgleason, 2018-02-07:
Is this it?
https://github.com/seandepagnier/weatherfax_pi/blob/master/appveyor.yml

I think Sean has to log into it and simply allow it to work.

rgleason, 2018-02-14:
Now when I go to the Code page , then the list of commits and on the most recent commit "yet more inlines"   ff01db2
https://github.com/seandepagnier/weatherfax_pi/commits/master

and hover over the green checkmark, I get green checkmarks for successful compilation of appveyor and travis ci.  Which is great.
[image]

When I go into the Appveyor "Details" link,  I can see the output of the entire compilation and the result.

```
    28 Warning(s)
    0 Error(s)
Time Elapsed 00:00:46.31
Collecting artifacts...
Found artifact 'build\weatherfax_pi-1.8.001-win32.exe' matching 'build\*.exe' path
Uploading artifacts...
[1/1] build\weatherfax_pi-1.8.001-win32.exe (713,996 bytes)...100%
"GitHub" deployment has been skipped as environment variable has not matched ("appveyor_repo_tag" is "false", should be "true")
Build success
```

So in order to download a copy of this file that has successfully compiled, the  "appveyor_repo_tag" needs to be changed to "true" or "1"
I find in the appveyor.yml file from did-g that
 
`line 68 appveyor_repo_tag: true # deploy on tag push only`

What should I do now to get the compiled file  "weatherfax_pi-1.8.001-win32.exe" in order to test it?
Note: since I do not have access to Sean's appveyor account, I will not be able to make any necessary changes to get the file saved. Do I have to fork and fetch locally, change the appveyor file so that it compiles using my appeveyor account, in order to get my hands on the compiled file?

nohal, 2018-02-14:
[image]

rgleason, 2018-02-14:
Yikes, that was quick. Thanks Pavel.   There are many layers/links.

rgleason, 2018-04-01:
mschiff, 
Here is a tagged release.
https://github.com/rgleason/weatherfax_pi/releases
can we close this now please?

rgleason, 2019-04-19:
To push a tag  (done after a commit).  
When on the branch and commit you want...
Examples
    git tag v1.9.2-ov42

    git push origin tag v1.9.2-ov42       to push the tag to remote 'origin'.

    git tag -d v1.9.2-ov2         to delete the local tag

    git push origin --delete <tagname>  to delete the remote tag

Perhaps we can close this now? 
It's old.

rgleason, 2022-04-22:
@seandepagnier   Please close this it is old and a distraction.

rgleason, 2022-05-22:
This is getting long in the tooth. Sean please close. OP does not respond. It is now a distraction. Thanks.

closed 2022-05-22 by seandepagnier

mschiff, 2022-05-24:
Sorry, yes, this can be closed, thanks