seandepagnier/weatherfax_pi #25: Open file dialog Lat/Lon inout fields are too small.

TR-Martin, 2016-04-26:
Sean,
 I'm not sure if this is the proper place for this issue report. but I couldn't find any other place. If I'm in the wrong here let me know. 

I wanted to bring to your attention an issue I've been bumping into with the OpenCPN WeatherFax. 

From a coding stand point it should be pretty simple to address, from a UX stand point it's making life difficult.

When I open a local weather fax image to overlay and it requests the image coordinates, the entry fields for the Lat and Lon are extremely small/crowded. So small in fact, that you cannot see the whole numbers. 

Before I go too far. I need to explain my setup, I'm running Windows 7 on a MacBook Pro with Retina display. There is a known issue with Windows 7 handling High DPI displays properly. which I think is contributing to this issue. As high DPI displays become more common place this issue might start affecting more people.

I've attached a screen capture to show you what I'm talking about..
Once gain if I'm in the wrong place for bringing up this issue please let me know.

Cheers, 
Todd Martin

P.S. I really like the plug in. it's very handy on the boat. 
[image]

rgleason, 2016-04-27:
You can change the screen resolution for the time being. http://windows.microsoft.com/en-us/windows-vista/change-screen-resolution

Also change font and icon size http://www.techtalkz.com/windows-7/516049-how-change-font-size-icon-size-windows-7-a.html  may help.

Hopefully you will not need to reenter lat/long coordinate information when the plugin starts automatically saving KAP files.  Why don't you try saving/exporting your files locally as KAP files which should have lat/long saved.  Then when you wish to refer to them again, load the Kap file instead off the image files.

TR-Martin, 2016-04-27:
Thanks for the reply Rick,
So I've already set the Font & Icon size to "Custom 200%" it's the max a user can set it to. (went through that exercise with Photoshop) So that base is covered. (and no upgrade/downgrading to Windows 10 is not an option, I think Microsoft has gone from spyware to down right creepyware)

Reducing the resolution, yea I can do that, but it's like asking a car dealer to put a 6 cylinder engine in a Corvette so there's room for another cup holder. I really want to keep the resolution up to maximize map information.

As for the KAP file, that may be workable, I originally came at this using a RTL-SDR dongle as a SSB radio replacement with SDR# and Seatty to capture the weather faxes. I'm in the middle of New Orleans' broadcast at the moment using WeatherFax for the first time to retrieve the images I'll get back to you when it's done broadcasting to see if using .KAP  files side steps this whole issue.

Cheers,
Todd Martin

rgleason, 2016-04-27:
Todd
I have reduced the resolution on my win10 because of text size. Just did it a little and it helped and awas faster.
This does need to be fixed and Pavel knows abouy it. I will make a Tracker  entry about it in the program too.
One other thing you could do is increase the font size to see ir that woulf help.I think that is under plugin preferences or ocpn options UI. Very interested in your fax experiences.

TR-Martin, 2016-04-27:
Rick,
Cool, Great to hear someone is looking into a root solution. If a tester is needed drop me a message I'd be more than happy to help validate.

On the font size suggestion I tinkered with changing the Font size under OpenCPN->options->user-interface->fonts I tried changing the font size for  the following:

- Chart Texts
- Console Value
- Dialog
- Menu
- ToolTips
- Status Bar
Unfortunately none of them had any affect on the Lat/Lon coordinate dialog box in WeatherFax, Worth a try though. 

As for my first fax experience: over all it was very positive. There was one deal breaker for me though. While it did decode the faxes, it did not automatically handle the APT start/stop signal between each weather fax page.  While I'm sailing I need unattended decoding, separating each fax into it's own page as it goes.(something Seatty handles for me) The great thing I did see was that your decoding algorithm seems to be about 10% more efficient than Seatty. (well task manager was showing an approximate 10% decrease in CPU load between Seatty and WeatherFax). 

I hope this all helps. It's a great tool, the ability to overlay weather faxes on OpenCPN makes it an extremely helpful tool in route planning.

Cheers,
Todd Martin

rgleason, 2016-04-29:
Sorry, I guess the font size is set in the plugin.  I will make a tracker request to expose the setting and also suggest  automatically handling the APT start/stop signal between each weather fax page.

seandepagnier, 2016-05-09:
Gah,

That is horrible.

It looks completely normal on any system I've ever seen so it must
have something to do with the sizers in wxwidgets.

It's probably easy to correct but I'm not sure how to go about doing
it without being able to test.

Do you see any other similar problems in other windows, or in any other plugins?

Also, are you decoding from audio with the plugin?  If so, did you
record or is it working well to use the plugin directly?

Sean

TR-Martin, 2016-05-09:
Sean,
     Are these questions for me or someone else?

Cheers,
Todd Martin

rgleason, 2016-05-10:
Todd
Questions for you.

TR-Martin, 2016-05-10:
Fair enough,
If WXwdigets is getting the fundamental screen geometries from Windows that could the source of the issue. Like I stated above Windows 7 has known issues with High DPI displays. 

**Other programs with a similar issue**
I have not seen another program with this exact issue but I do have an issue with Adobe Photoshop where the interface is extremely small. Adobe is is pointing the finger at MS for not reporting the screen geometry properly. MS is saying they fixed it in Windows >8.0 and if you want it fixed you have to upgrade. Not sure what to make of that.?
_(As a side note I don't think I have any other programs using WXwidgets.)_

**OpenCPN **
I just noticed something... your question got me looking at OpenCPN and it's interface. OpenCPN does not have numerical inputs with support for "wxSP_ARROW_KEYS:" the closest I could find was the options->charts->Vector Chart Display-> Depth. (see attachment below) there is some clipping but it's not very bad, which I think is due to the fact the arrow icons are not being displayed in the input field. 

**Decoding: **
I've been using SRD# -> virtual audio cable -> SeaTTY to capture the fax pages then using weather fax to overlay them on to OpenCPN. I tried using Weatherfax to capture the images but it did not honor the APT start/stop signals - I ended up with a single image made up of 8 pages of faxes. I'm usually single handing on the boat and I need the fax decode to autonomously retrieve and save  each page individually. (SeaTTY does this) so the short answer is "Not right now, wetherFax is not capturing the faxes for me."  

**Debugging**
My offer still stands, since I have a known system to reproduce the issue on, I have no problem playing the part of validation technician. Granted it would be a somewhat slow process: emails and screen shots but in the end it would work. Totally up to you.

Cheers,
Todd Martin

[image]

rgleason, 2016-08-11:
NOTE: Sean, I have been using the older Opencpn v4.2  version of Weatherfax_pi because the current on github your repository does not have the Weatherfax Wizard working, the images come up blank.

So my comments here are related to how image wizard appears in the version 5001 that pavel put up on the opencpn.org website.    Image wizard does not work in the current github version!

I am finding many of the same problems with values being cut off etc. I will post a couple of shots.

When first open it. Have to Stretch right, stretch down, stretch up, strech column, 
Would like it to open so menu items show, and when I stretch and adjust it, it remembers what the setting is, so I don't have to do it each time I use weatherfax.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

rgleason, 2016-08-25:
The LAT and LONG entry areas are now wider and visible.  What a relief! 
Closing

rgleason, 2016-08-25:
No I can't close it.  TR-Martin has to, or Sean.

seandepagnier, 2016-08-25:
When did they become wider?

I noticed latest github of wxwidgets now uses pixels per inch of the screen to calculate sizers, but you have to rebuild wxwidgets from source, not sure if it's in 3.1 or not.

rgleason, 2016-08-26:
The size changed just now with the current version on my 1920 x 1060 screen, I think. The older one I have been using the image is smaller, sometimes too small to pick corners right, or decipher the lat long.  But this is on my 2160 x 1440 screen too.

I can deal with this, I think. Don't worry about it.

TR-Martin, 2016-08-29:
Ooops sorry everyone I just saw the emails on this..I've got my system reconfigured and updated  I'll test out the new code after 7:00pn CST (next Wefax broadcast from NOLA) if it's all good I'll close the issue.. thanks for the support...

rgleason, 2016-08-30:
TR. when you  check it please see if there is a memory leak which makes opencpn freeze up and bomb while downloading files. In windows it ionly takes 1 to 4 file downloads to happen. Thanks

rgleason, 2016-08-30:
So I still must use the older version.

rgleason, 2016-09-01:
TR What is your result?

rgleason, 2016-09-05:
TR this should be closed now it is fixed.

rgleason, 2016-09-05:
Howevwr mem leak is,xovered elsewhere

TR-Martin, 2016-09-06:
Ok I'm back from the holiday.. it took me a while to get everything setup and working... I did not see any lock ups or crashes that might suggest a memory leak. however the choice boxes look much better enough that I'm going to go ahead and close this issue. 
Thanks for the help, much appreciated...

Cheers.
Todd Martin

closed 2016-09-06 by TR-Martin

rgleason, 2016-09-06:
TR 
Good you closed this now, but can you test you're linux version as I have with Windows?
See the bottom post here please  https://github.com/seandepagnier/weatherfax_pi/issues/38
Thanks.