seandepagnier/weatherfax_pi #30: Partially hidden buttons in Internet Retrieval Dialog

nkiesel, 2016-08-06:
The 2 buttons "Retrieve Scheduled" and "Retrieve Selected" are often partially or completely hidden when the window is shown for the first time.  They can be made visible by adjusting the slider between the file selector and the button panel, but of course the user has to know that they are there to begin with (and that she can adjust the slider).  I tried fixing this using a different value for the 3rd. parameter of `SplitHorizontally`, but that did not work.

This happens both under Linux and under Windows.

Also I think it would be nicer if these 2 button would be in a row of their own under the "Contains", "Server", "Regions" selectors because they do not belong to "Contains", but that is a more minor point.

[image]

[image]

seandepagnier, 2016-08-06:
> ![screenshot from 2016-08-06
> 01-05-45](https://cloud.githubusercontent.com/assets/1134209/17455641/60353ca4-5b73-11e6-809f-e3889a8da3fb.png)
> 01-19-09](https://cloud.githubusercontent.com/assets/1134209/17455656/d21422e0-5b73-11e6-8cb2-e0d21f6bab3d.png)

The buttons are in a row of their own outside the contains box.

What is your screen size?

nkiesel, 2016-08-06:
Screen size is 1920 x 1080, and the Windows laptop (which was running 4.2)
had the same resolution I think.

rgleason, 2016-08-10:
Same problem on a SurfacePro3  with 2110 x ?  Resolution

seandepagnier, 2016-08-10:
What about the All and None buttons?

rgleason, 2016-08-11:
Sean
I'll try some screenshots. All and None are okay once you stretch things out, but they are hidden each time you start,  and each time you must stretch it out in a number of ways and it is not remembered either.. Darn the screenshots are on the other computer. I will upload those tomorrow morning. Sorry about that.

rgleason, 2016-08-25:
Sean, the new version which I just built from your today's version is better with regard to saving settings, and it does at least open with the menus below just a bit more visible, however the  Retrieve Scheduled and Retrieve Selected are still hidden.

I can stretch the bottom panel up by pausing for the arrow and use the buttons, but this does not seem to be remembered after shutdown of O.
  --Yes, I know this wxformbuilder interface is a pain to work with. I have experienced it and so has Thomas R.

Ideas for Internet Reetrieval Menu
1.   I wonder if the space around the all buttons could be tightened up a little so that the entire menu could get smaller?
2. Could Lat and Long entries be horizontal to save space? 
3. Initially, could the entire bottom menu with the controls, servers and regions open with just 3-4 lines of retrieval?  
4.Then user could set the height and it would be saved?

seandepagnier, 2016-08-26:
all 4 ideas are possible.

I don't have any problem finding the buttons.  I will play some more with it.

Sean

rgleason, 2016-08-26:
Here is two other ideas for reducing real estate and making the buttons smaller, Its a more vertical format that takes the Text out of the button.

Retrieve
Scheduled  [X}    <--click on the button
Selected     [X]     <--click on the button

Retrieve
[Scheduled]  <--click on the button
[Selected]     <--click on the button

rgleason, 2016-08-26:
Maybe you could take this vertical arrangement and put it to the left of the  Retrieve

Contains                Retrieve              Servers        Regions
[Boat Position]       [Scheduled]        List               List
Lat     [       ]            [Selected]         
Long  [       ]
[Reset]                                              [All] [None]   [All] [None]

It might be easier to make FormBuilder to behave with 4 boxes across.

rgleason, 2016-08-26:
This computer display is 1920 x 1080 the  other one is  2160 x  1440

rgleason, 2017-02-16:
This is still a problem because the plugin window does not open with the buttons showing. The plugin window has to be dragged out in three different ways to get the full view. This should be fixed.

rgleason, 2017-02-16:
A. Make the box around these buttons tighter/smaller.
B. Make all the buttons show when the menu is opened, with just 3 lines above, for a start.      
[Boat Position]
[Reset]
[Retrieve Scheduled] [Retrieve All] [All]  [None}

OR maybe use    [Get Scheduled]  [Get All]    [All] [None]

rgleason, 2017-02-18:
I'm going to add some screenshots here to illustrate.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

rgleason, 2017-02-18:
-----
**Image Coords operation and space!**
Top one shows Image Coordinates.
The operation of the Image Coordinates needs a little work, as it is not entirely clear how it works.
First there is only a choice of "New Coord",  then if you reload the Image Wizard, the "New Coord" turns into New Coord 1, New Coord2,  etc.   Why don't these all show as selections when you first drop down the menu?
-----
**Name your own  Coords**
Also I thought you could name your own coord so that you could determine what those coords apply to later when you are editing weatherfaxInternetRetrieval and CoordinateSets.xml?    It does not seem to work that way.  It would be very nice.
-----

**Restricted LAT/LON space**
The additional important thing the is screenshoot shows is the restricted space for LAT/LONG. This is very very inconvenient and hard to work with. It makes it impossible to determine if the settings you selected were entered properly, which is often "rejected" or different, leading to overlay problems.
Also fields at the bottom are a little restricted too.
------

**Unnecessary Stretching**
The bottom 4 shots show how much stretching is required each and every time we open the plugin!!!
It takes 3 separate stretches to get setup, each and every time!  Why can't this be remembered? or even figured out while opening the menu???

Additionally it is hard to get the middle stretch one into stretch mode. It is sensitive.
------

**Reorganize the Fields for easier Use**

------- Contains   -------------------Servers -----------Regions
[Get Scheduled]  [Get Selected] 
Lat [----------]
Lon [---------]
[Boat Position] [Reseet]
<- Try to reduce the width-->

We use Get Scheduled & Get Selected most often, it is hidden all the time and requires stretching.

rgleason, 2017-07-31:
Sean, I know how formatting screens is a Pita and how frustrating it can get. It would be very helpful if somehow when the first weatherfax internet retrieval screen is opened, that the "Retrieve Selected" and "Retrieve Scheduled" buttons were to be showing at the start!  

 Otherwise new users will never realize that those buttons are used to download the files!
This is very important and is probably why the plugin is not more popular than it already is!

If this could be fixed somehow that would be an accomplishment!
Best
Rick

seandepagnier, 2017-07-31:
wow, this is something I didn't realize.
I'll see what I can do.  wxwidgets is really frustrating.

Sean

rgleason, 2017-08-22:
Sean, While starting to compile and build Stelian's version of wxfax with the scripts, I had an idea how to fix the disappearing menu problem, by moving the selection menus up above the listings. I think it is still a consistent interface and it might be even better for new users because hopefully the Main selection and execution menus will always show.

Of course if the user defined sizing and location of this menu were remembered between sessions that would help too.

[image]

rgleason, 2017-09-12:
Please Merge this Pull Request which fixes the problem.
https://github.com/rgleason/weatherfax_pi/pull/4

rgleason, 2017-09-12:
Made the pull request to Sean's Repository\\
https://github.com/seandepagnier/weatherfax_pi/pull/54/commits

rgleason, 2017-09-13:
Merged. Thank you Sean. I think this can be closed now. I can't do it. Sean or kneisel can close it.

rgleason, 2017-09-14:
Well PR #54 was wrong, all the files were in the root directory.  I've made three separate PR #55, 56, 57 to fix this and to update the XML files as of today.  
PR 55  Fix the InternetRetrieval Menu
PR 56 , Fix &Update XML & README files
PR 57   Fix the Windows Preferences Bug

In PR 56 added instruction to Windows users to delete the old  
programdata/opencpn/plugins/weatherfax 
coordinatesets.xml
usercoordinates.xml files
 for a "clean" start.

Here is what the InternetRetrieval Menu looks like when opened.  Scheduled and Selected are accessible.
[image]

rgleason, 2017-10-15:
I have fixed this in my version of the plugin by moving the menus and adusting things.

rgleason, 2017-10-15:
When PR #68  is Merged this will be fixed.

rgleason, 2017-11-10:
#71 has been merged.
This can be closed now. 
I can't close it. Only Sean or  Nkiessel can close it.

nkiesel, 2017-11-10:
Thanks for all the hard work!

closed 2017-11-10 by nkiesel