seandepagnier/weatherfax_pi #34: User Update of Data files - coordinatesets.xml and weatherinternetretrival.xml

rgleason, 2016-08-25:
8/25/2016 Completed NOAA sections  (several satellite need work)
8/25/2016  Made OPC section
8/25/2015  Completed NOAA OPC section
A number of other sections need updates.

rgleason, 2016-08-25:
With the version current today, I see now that NOAA  Pt.Reyes  first 7 have to be corrected. Also there are some others under NOAA other regions.

There are some problems with downloading now. The older version works much better with regard to downloading.  Noted in sep Item.

rgleason, 2016-08-25:
Also, it there are a lot of images popping into Image Wizard, that should not be because those coordinates were all set properly, and they are working on my other machine with an earlier version of Opencpn 4.2 and earlier version of Weatherfax.

Did the Pull Request files go through without change?
What could be causing this problem?

rgleason, 2016-08-25:
I wonder if I should have included my usercoordinates.xml file or something...?

rgleason, 2016-08-25:
Programdata/opencp/plugins/weatherfax/data  did have old versions in it and I deleted both coordinatesets.xml and usercoordinates.xml

The first time I opened Opencpn afterwards, and started downloading until it failed again pretty quickly, I did not see and coordinatesets.xml or usercoordinates.xml in the programdata/opencpn/plugins/weatherfax/data directory.

The second time I opened, after the failure, I found those files.

However I am still getting a spinning wheel and then eventually failure when trying to download a file, I have not noticed a pattern with respect to failures.   Something is wrong though, because the other computer downloads reliably.

rgleason, 2016-08-25:
PS I think the user should be asked to delete the programdata/opencpn/plugin/weatherfax/data files
unless these files are coming from your machine somehow.

seandepagnier, 2016-08-26:
Rick,

Sorry yes.  If your local versions don't exist it uses the plugin
version.  From there on it will use the local version unless you
delete it.

So in the case that you download a new version from github you have to
delete the local version.   It should compare timestamps and prompt
the user if they are newer.  It should really "merge" them so as not
to delete coordinates the user entered.

Maybe it should maintain both sets, and prefer the user's coordinates
over the builtin ones?  Maybe coordinates which also have a valid url
should be in a separate file from ones without a url?

As a possible feature it could download directly from github and
update the coordinate sets.

Sean

rgleason, 2016-08-26:
This is a really complex subject. I am sorry we ran against this. 
Pavel has a website for Chartdownloader catalogue updates, maybe that model would help?
Maybe Pavel has some good solutions to this problem which would also allow us to update the XML's or catalogues fairly easily?  I think Transmitter Dan and Pavel update the website catalogues when needed with some password access and then users first update catalogues and then select for download.
The weatherfax could be similar maybe.

rgleason, 2016-08-26:
For the time being, I think we could ask users to delete the three files in 
c:/programdata/opencpn/plugins/weatherfax/data/ ?
Presuming that the new XML files are good.

Also a plea that users submit their changes has been made, and a notice the new plugin will ovewrite their user defined coordinates and they should save them or share them with others via the forum.

rgleason, 2016-08-26:
Invalid urls  
Maybe the program should add [invalid url] to weatherfaxInternetRetrieval.xml lines?
It would help the person updating, and after a second test, a similar message would be good to have when the user tries to download.  Graceful failure, a message and then recovery would be best.

Then perhaps these lines are sent back to a server log?

Invalid coords
Maybe we when the user has to set the coords and it comes up in the Image Wizard, at the end of the process, after the image has been overlaid, there is a question to the user, 
Is this image ok? Yes No.  If yes then those coords and the image file name is sent to a server log.

From those logs someone updates the XML files in the server.

Just an idea.

rgleason, 2017-02-16:
I have some files somewhere that are more current.
 weatherfaxInternetRetrieval.xml
coordinatesets.xml 

and will try to find them. Meanwhile the current files from Aug 15 2016 have some problems....
I might have to make another update of the more current files from 4 months ago.

rgleason, 2017-02-18:
Please see the CF Forum Thread Weatherfax  http://www.cruisersforum.com/forums/f134/weatherfax-97533.html#post2328606

I have updated the WxFax InternetRetrieval Files.  This is a very comprehensive test and update started last summer and just finished. There are still a few weatherfaxes which need coordinatesets.xml record improvements, and we are missing the URL for several important websites, but there are about 3-4 new weatherfax URL's and the files are comprehensive in the sense that they have all been tested, fixed, URL's changed where found, etc. 

Please review the readme file and help with improved coordinateSet xml records where possible.
Unzip the file and install as explained in the readme.

Thank you
[WeatherFax_pi-Update-Internet-Retrieval-2-18-2017 - Copy.zip](https://github.com/seandepagnier/weatherfax_pi/files/785299/WeatherFax_pi-Update-Internet-Retrieval-2-18-2017.-.Copy.zip)

rgleason, 2017-02-27:
PR#34  has been Merged, so this has been done.  Closing.

closed 2017-02-27 by rgleason