seandepagnier/weatherfax_pi #35: 10-25-2016 Version - Download Problems - Hangs up if there is any problem connecting.

rgleason, 2016-08-25:
I was having some trouble after trying todays version. 
For example NOAA Boston 96hr Surface Forcast would not download and caused the entire program to go into a tailspin and close.  Failure should be much more gentle than that I think.

This is just one example, I have numerous crashes and yet I was able to download and display faxes successfully, but it is tedious to have to restart the program etc.  

I have checked norton firewall and all opencpn.exe files are "Allowed"  So that is not the problem.

There did not seem to be any rhym or reason to the ones that don't download and cause Opencpn to crash.   I was using Opencpn v4.4  and compiled weatherfax with opencpn.lib for version 4.4.

seandepagnier, 2016-08-25:
Was the download dialog visible when it crashed?  Isn't there an abort button?

rgleason, 2016-08-26:
Yes the the dialog is there momentarily but then it disappears, the round blue busy appears, the screen turns black and it freezes.  and in task manager memory use by O goes up to 1717 mb

rgleason, 2016-08-26:
Yes, I can confirm that there is a memory leak. Task manager goes up each time I download, until it hits a max and freezes.

rgleason, 2016-09-16:
See the memory leak issue. This is being closed.

closed 2016-09-16 by rgleason