seandepagnier/weatherfax_pi #36: Image Wizard - helpful if user had some scale slider for the image.

rgleason, 2016-08-25:
It would enable picking better coords if image is too small, and would help if image comes in too big.

This is just an idea, it should not interfere with the scaling of the image to the charts, it would just be for use in the Image Wizard.

seandepagnier, 2016-08-26:
maybe a screen magnifier program can work?   This feature is possible
but very low priority.

rgleason, 2016-08-26:
It would help with picking coords and sizing the image so we don't have to scroll as much.
Understand it is low priority

rgleason, 2017-02-18:
Also the weatherfaxes are terrrible at properly labeling coordinates, and in fact the coordinates can be just wrong!!  Being able to zoom in and see the text and notes for coordinates would be very helpful.

It won't help at all if the coords are non existant as in some cases where we can only do trial and error to fit.

rgleason, 2018-01-19:
One reason a scale slider is needed is so that we can inspect and identify the coordinates which are always too small and difficult to read.  I suppose a magnifying glass could also be used, but sometimes we need to zoom into the image to discern the coordinates.

rgleason, 2018-01-23:
One workaround is to get the polars to just work somehow by guessing the coords and to complete the wizard any way possible so that you can zoom into the overlay and decipher the proper coordinates.