seandepagnier/weatherfax_pi #37: Image Wizard - after completion - ability to drag the image intact to fit a shoreline?

rgleason, 2016-08-25:
Sometimes a mercator image does not have good coord markings, and we get an image that is offset. It might help if we could just move the image over to the shoreline somehow. I think this happens when the weather station making the faxes does not use exactly the same coordinates or has changed them. The image is basically ok, but needs to be moved a little.

Just an idea.  I could be misused I suppose.

seandepagnier, 2016-08-26:
Yes, I intend to support dragging the corners, edges or entire image
to edit the image directly on the chart.   This can be used to correct
charts as well, but the plugin would read kap files to do it.

rgleason, 2016-08-26:
Sounds very good to me.  Should I close this now?

rgleason, 2016-09-16:
I guess I'll close this now Sean, as  you have it in mind.

closed 2016-09-16 by rgleason