seandepagnier/weatherfax_pi #38: Memory Leak

rgleason, 2016-08-30:
Sean,
I have a memory leak in the current version. Is there some possibility I did not compile it right?
It bombs between 1 and 4 image downloads.

rgleason, 2016-08-31:
Sean, I tried it again several times and this time took a screenshot.

[image]

rgleason, 2016-08-31:
I am going to try compiling again and try to make sure I have all the necessary files, but are you aware of any download files that might have changed that Windows does not normally have?

Perhaps download techniques changed??

rgleason, 2016-09-01:
Build succeeded.

"C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\ALL_BUILD.vcxproj" (default target)
(1) ->
"C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj" (default targ
et) (3) ->
(ClCompile target) ->
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\DecoderOptionsDialog.cpp(47): warning
 C4800: 'long' : forcing value to bool 'true' or 'false' (performance warning) [C:\Users\Rick\Docum
ents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\DecoderOptionsDialog.cpp(48): warning
 C4800: 'long' : forcing value to bool 'true' or 'false' (performance warning) [C:\Users\Rick\Docum
ents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\inttypes.h(126): warning C4005: 'P
RId64' : macro redefinition [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherf
ax_pi.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\inttypes.h(132): warning C4005: 'P
RIi64' : macro redefinition [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherf
ax_pi.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\algorithm(1026): warning C4996: 's
td::_Transform1': Function call with parameters that may be unsafe - this call relies on the caller
 to check that the passed values are correct. To disable this warning, use -D_SCL_SECURE_NO_WARNING
S. See documentation on how to use Visual C++ 'Checked Iterators' [C:\Users\Rick\Documents\GitHub\o
-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\algorithm(1026): warning C4996: 's
td::_Transform1': Function call with parameters that may be unsafe - this call relies on the caller
 to check that the passed values are correct. To disable this warning, use -D_SCL_SECURE_NO_WARNING
S. See documentation on how to use Visual C++ 'Checked Iterators' [C:\Users\Rick\Documents\GitHub\o
-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\xutility(2132): warning C4996: 'st
d::_Copy_impl': Function call with parameters that may be unsafe - this call relies on the caller t
o check that the passed values are correct. To disable this warning, use -D_SCL_SECURE_NO_WARNINGS.
 See documentation on how to use Visual C++ 'Checked Iterators' [C:\Users\Rick\Documents\GitHub\o-p
lugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\algorithm(1026): warning C4996: 's
td::_Transform1': Function call with parameters that may be unsafe - this call relies on the caller
 to check that the passed values are correct. To disable this warning, use -D_SCL_SECURE_NO_WARNING
S. See documentation on how to use Visual C++ 'Checked Iterators' [C:\Users\Rick\Documents\GitHub\o
-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\libaudiofile\aupv.c(112): warning C40
18: '<' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\w
eatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\libaudiofile\aupv.c(118): warning C40
18: '>' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\w
eatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\libaudiofile\aupv.c(132): warning C40
18: '<' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\w
eatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\libaudiofile\aupv.c(138): warning C40
18: '>' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\w
eatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\libaudiofile\aupv.c(152): warning C40
18: '<' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\w
eatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\libaudiofile\aupv.c(158): warning C40
18: '>' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\w
eatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\libaudiofile\aupv.c(187): warning C40
18: '<' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\w
eatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\libaudiofile\aupv.c(193): warning C40
18: '>' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\w
eatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\libaudiofile\aupv.c(207): warning C40
18: '<' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\w
eatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\libaudiofile\aupv.c(213): warning C40
18: '>' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\w
eatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\libaudiofile\aupv.c(227): warning C40
[... 1 lines cut ...]
eatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\libaudiofile\aupv.c(233): warning C40
[... cut ...]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\libaudiofile\extended.c(109): warning
 C4244: '=' : conversion from 'unsigned long' to 'unsigned char', possible loss of data [C:\Users\R
ick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\libaudiofile\extended.c(110): warning
 C4244: '=' : conversion from 'unsigned long' to 'unsigned char', possible loss of data [C:\Users\R
ick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]

    26 Warning(s)
    0 Error(s)

rgleason, 2016-09-04:
Memory leak seems to be worse when you download successive files, about 4 -5 in a row without waiting.
When you download a file and wait, then download a little later say 4-5 minutes wait, it seems to be ok.
I have been trying to get the debug version of the plugin working under MSVC++ so that I can give you better information. When I sort that out I'll advise.

seandepagnier, 2016-09-04:
it is or isn't possible to easily reproduce?

rgleason, 2016-09-04:
This morning I stress tested to see why its happening...

Sean, using  NOAA (Boston  & Pt Reyes)  and OPC Atlantic  and NOAA OPC (Atlantic & Pacific)
If I click on the top twice to select all files then try  "Retrieve Scheduled"  when it is actually working (This button works on the NOAA (Boston & Pt Reyes)  OR with the others I simply highlight all the files and hit "Retrieve Selected"   (maybe it should say "Retrieve Highlighted"?)   and they generally download fast (This is on a good internet connection)  - There are some links I need to fix still.

However once files are downloaded and you do this again, there is a Prompt which comes up...
"Fax downloaded less than 180 minutes ago. Use Existing File?  Yes  No"
Each time I hit "Yes" it seems to increase OpenCPN's memory use.  It generally seems to be about 10-60mb, but then sometimes there is no increase when you hit "Yes". This is puzzling.

I then thought that if I hit "No" the memory use would stay the same, no, it jumps up about 100mb, then if I hit "No" each time it jumps 20-50mb but sometimes it does not jump at all.

When the memory use for Opencpn gets up to about 1700mb then we start getting freezeups and misbehaving.

I'll test some more when I get a chance and I think I need to fix a few links in the ones I've done and go ahead and confirm and fix the other Fax retrieval links too.

rgleason, 2016-10-26:
I am closing this. Will open a new one.

closed 2016-10-26 by rgleason