seandepagnier/weatherfax_pi #39: memory leak

rgleason, 2016-10-26:
After automatically retrieving many images, the opencpn memory creepsup to something like 1600mb and then weatherfax starts to choke misbehave and freeze.

rgleason, 2017-02-16:
Just bombed out twice due to continued memory leak.  The second time was very serious, with an urgent message from Windows to close programs right away due to lack of memory. I then bombed and it took awhile to reboot because the SurfPro3 had gotten so hot.

rgleason, 2017-07-30:
Still a problem. The new UK-Met weatherfaxes provide a good example. Downloaded they are 50kb-70kb, but each one when downloaded increases Opencpn memory use by 110mb!

See this report http://www.cruisersforum.com/forums/f134/weatherfax-97533.html#post2444741

rgleason, 2017-07-30:
We have determined that every time a 50kb-70kb weatherfax is loaded, the Opencpn memory use increases by about 110mb!   When weatherfax pi is closed the memory is not recovered.  When the weatherfax files are deleted, the memory is recovered. If Opencpn is closed and restarted then the memory is recovered.

It seems that the weatherfax_pi is opening files and not closing them when weatherfax is closed.
It also seems that weatherfax_pi is using an awful lot of memory to display 50kb-70kb files even if they must be recalculated with coordinatesets.xml information to be overlayed. 

We believe this memory situation which can lead to OpenCPN crashes if many weatherfax files are opened, can be improved with a few small file management improvements.
1. Close all weatherfax files when the plugin is closed.
2. See if there is some way to improve the projected or overlaid file size.

Other users, including Linux have noticed this.

seandepagnier, 2017-07-31:
I know it stores 3 copies of the image in ram, plus another copy in
video memory.

This is uncompressed, so the same as if the image were a bitmap (32
bit)...   A 57k png file, could uncompressed to several megabytes.
Not sure how 4x this can actually reach 110mb though, that is
surprising.    The image would need to be 2600x2600 pixels, is this
what you have?

It could be smarter and use only the copy in video memory. It could
also use compressed textures

It could therefore use 1/24th the memory it currently uses, but my
head is in autopilot space.

rgleason, 2017-08-01:
Yes, Sean. I understand you are doing pilot. Its not urgent, just for consideration later. Thanks for advising.

rgleason, 2018-01-29:
This memory leak still exists, but it is much better due to Sean and Did-g.
At some point the memory handling will have to be improved, but for now I am closing this.

closed 2018-01-29 by rgleason