seandepagnier/weatherfax_pi #40: NOAA URL for download changed

WxJon, 2016-11-21:
WeatherFax 1.5001 points to http://weather.noaa.gov/pub/fax/  .. should point to http://tgftp.nws.noaa.gov/

See NWS Change Notice 16-16 at http://www.nws.noaa.gov/om/notification/scn16-16wngccb.htm

rgleason, 2016-11-22:
I have updated all the links and prevouusly uploaded those xml.
I have also updated and tested all xml links but have not yet uploaded due to ongoing changes by noaa and others. Sometime in the next month I will check all the links again, modify as needed and make a pull request. If you want the partial update look to cruiser forums weatherfax thread where the xml files have been attached to a post.
If you want to help this effort provide corrected xml files of yourarea.

rgleason, 2017-02-16:
Yes this can be closed I believe.

rgleason, 2017-02-18:
This is fixed by the most recent files, available here.
https://github.com/seandepagnier/weatherfax_pi/issues/34

This issue can be closed.

rgleason, 2017-02-27:
WxJon or Sean,  please close this Issue as it is fixed by recent PR #34    Only Sean and OP WxJon can close this. Thanks.

WxJon, 2017-02-27:
Thanks for timely action.

closed 2017-02-27 by WxJon