seandepagnier/weatherfax_pi #45: New git clone  - 20 Errors

rgleason, 2017-03-05:
Sean,
I tried with earlier setup git fetch  and then deleted it.
Then did git clone  and copied all files to the source\buildwin directory
1. 2 portaudio
2. pvw32con.exe
3. all the new buildwin files with crashrpt, vc, expat, gtk, wxWidgets etc for svg icons.
Then copied version 4.2 of opencpn.lib to new source\build
Then from source\build   executed cmake -T ..   and cmake --build .
These are the errors.

------------------
  Generating Code...
Done Building Project "C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfa
x_pi\build\weatherfax_pi.vcxproj" (default targets) -- FAILED.

Done Building Project "C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfa
x_pi\build\ALL_BUILD.vcxproj" (default targets) -- FAILED.

Build FAILED.

"C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\ALL_BUILD.v
cxproj" (default target) (1) ->
"C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_
pi.vcxproj" (default target) (3) ->
(ClCompile target) ->
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(350): error C2511: 'WeatherFaxWizard *WeatherFax::OpenWav(wxString,AFframeco
unt,wxString,wxString,wxString)' : overloaded member function not found in 'Wea
therFax' [C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\we
atherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(351): error C2227: left of '->GetValue' must point to class/struct/union/gen
eric type [C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\w
eatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(352): error C2227: left of '->GetValue' must point to class/struct/union/gen
eric type [C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\w
eatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(353): error C2227: left of '->GetValue' must point to class/struct/union/gen
eric type [C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\w
eatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(358): error C2228: left of '.GetCount' must have class/struct/union [C:\User
s\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxpr
oj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(359): error C2597: illegal reference to non-static member 'WeatherFax::m_Bui
ltinCoords' [C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build
\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(359): error C3867: 'WeatherFax::m_BuiltinCoords': function call missing argu
ment list; use '&WeatherFax::m_BuiltinCoords' to create a pointer to member [C:
\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.
vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(359): error C2109: subscript requires array or pointer type [C:\Users\Freder
ick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(359): error C2227: left of '->name' must point to class/struct/union/generic
 type [C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weath
erfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(360): error C2597: illegal reference to non-static member 'WeatherFax::m_Bui
ltinCoords' [C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build
\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(360): error C3867: 'WeatherFax::m_BuiltinCoords': function call missing argu
ment list; use '&WeatherFax::m_BuiltinCoords' to create a pointer to member [C:
\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.
vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(360): error C2109: subscript requires array or pointer type [C:\Users\Freder
ick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(362): error C2228: left of '.m_CaptureSettings' must have class/struct/union
 [C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax
_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(369): error C2597: illegal reference to non-static member 'WeatherFax::m_Use
rCoords' [C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\we
atherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(369): error C2440: 'initializing' : cannot convert from 'WeatherFaxImageCoor
dinateList WeatherFax::* ' to 'WeatherFaxImageCoordinateList *' [C:\Users\Frede
rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(371): error C2671: 'WeatherFax::OpenWav' : static member functions do not ha
ve 'this' pointers [C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_p
i\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(371): error C2661: 'WeatherFaxWizard::WeatherFaxWizard' : no overloaded func
tion takes 4 arguments [C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherf
ax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(379): error C2227: left of '->Enable' must point to class/struct/union/gener
ic type [C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\wea
therfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.c
pp(386): error C2228: left of '.push_back' must have class/struct/union [C:\Use
rs\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxp
roj]
  C:\Users\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\SchedulesDia
log.cpp(799): error C2039: 'contains' : is not a member of 'wxString' [C:\Users
\Frederick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxpro
j]

    0 Warning(s)
    20 Error(s)

Time Elapsed 00:00:10.18

rgleason, 2017-03-09:
Today I tried a git clone on my  Win10 Laptop using Opencpn.lib from v4.5.315 
The 20 errors are the same I think.

Build FAILED.

"C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\package.vcxproj" (default target) (1
) ->
"C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\ALL_BUILD.vcxproj" (default target)
(2) ->
"C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj" (default targ
et) (4) ->
(ClCompile target) ->
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(350): error C2511: 'We
atherFaxWizard *WeatherFax::OpenWav(wxString,AFframecount,wxString,wxString,wxString)' : overloaded
 member function not found in 'WeatherFax' [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi
\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(351): error C2227: lef
t of '->GetValue' must point to class/struct/union/generic type [C:\Users\Rick\Documents\GitHub\o-p
lugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(352): error C2227: lef
t of '->GetValue' must point to class/struct/union/generic type [C:\Users\Rick\Documents\GitHub\o-p
lugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(353): error C2227: lef
t of '->GetValue' must point to class/struct/union/generic type [C:\Users\Rick\Documents\GitHub\o-p
lugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(358): error C2228: lef
t of '.GetCount' must have class/struct/union [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax
_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(359): error C2597: ill
egal reference to non-static member 'WeatherFax::m_BuiltinCoords' [C:\Users\Rick\Documents\GitHub\o
-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(359): error C3867: 'We
atherFax::m_BuiltinCoords': function call missing argument list; use '&WeatherFax::m_BuiltinCoords'
 to create a pointer to member [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weath
erfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(359): error C2109: sub
script requires array or pointer type [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\buil
d\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(359): error C2227: lef
t of '->name' must point to class/struct/union/generic type [C:\Users\Rick\Documents\GitHub\o-plugi
n\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(360): error C2597: ill
egal reference to non-static member 'WeatherFax::m_BuiltinCoords' [C:\Users\Rick\Documents\GitHub\o
-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(360): error C3867: 'We
atherFax::m_BuiltinCoords': function call missing argument list; use '&WeatherFax::m_BuiltinCoords'
 to create a pointer to member [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weath
erfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(360): error C2109: sub
script requires array or pointer type [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\buil
d\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(362): error C2228: lef
t of '.m_CaptureSettings' must have class/struct/union [C:\Users\Rick\Documents\GitHub\o-plugin\s-w
eatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(369): error C2597: ill
egal reference to non-static member 'WeatherFax::m_UserCoords' [C:\Users\Rick\Documents\GitHub\o-pl
ugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(369): error C2440: 'in
itializing' : cannot convert from 'WeatherFaxImageCoordinateList WeatherFax::* ' to 'WeatherFaxImag
eCoordinateList *' [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcx
proj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(371): error C2671: 'We
atherFax::OpenWav' : static member functions do not have 'this' pointers [C:\Users\Rick\Documents\G
itHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(371): error C2661: 'We
atherFaxWizard::WeatherFaxWizard' : no overloaded function takes 4 arguments [C:\Users\Rick\Documen
ts\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(379): error C2227: lef
t of '->Enable' must point to class/struct/union/generic type [C:\Users\Rick\Documents\GitHub\o-plu
gin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(386): error C2228: lef
t of '.push_back' must have class/struct/union [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfa
x_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\SchedulesDialog.cpp(799): error C2039
: 'contains' : is not a member of 'wxString' [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_
pi\build\weatherfax_pi.vcxproj]

    0 Warning(s)
    20 Error(s)

rgleason, 2017-03-19:
Sean - What Version of Opencpn.lib should we be using?

rgleason, 2017-03-19:
New git clone using Opencpn.lib from version 4.5  fails  same 20 errors.
Put required binaries in buildwin.

Done Building Project "C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\package.vcxpro
j" (default targets) -- FAILED.
Build FAILED.
"C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\package.vcxproj" (default target) (1
) ->
"C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\ALL_BUILD.vcxproj" (default target)
(2) ->
"C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj" (default targ
et) (4) ->
(ClCompile target) ->
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(350): error C2511: 'We
atherFaxWizard *WeatherFax::OpenWav(wxString,AFframecount,wxString,wxString,wxString)' : overloaded
 member function not found in 'WeatherFax' [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi
\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(351): error C2227: lef
t of '->GetValue' must point to class/struct/union/generic type [C:\Users\Rick\Documents\GitHub\o-p
lugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(352): error C2227: lef
t of '->GetValue' must point to class/struct/union/generic type [C:\Users\Rick\Documents\GitHub\o-p
lugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(353): error C2227: lef
t of '->GetValue' must point to class/struct/union/generic type [C:\Users\Rick\Documents\GitHub\o-p
lugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(358): error C2228: lef
t of '.GetCount' must have class/struct/union [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax
_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(359): error C2597: ill
egal reference to non-static member 'WeatherFax::m_BuiltinCoords' [C:\Users\Rick\Documents\GitHub\o
-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(359): error C3867: 'We
atherFax::m_BuiltinCoords': function call missing argument list; use '&WeatherFax::m_BuiltinCoords'
 to create a pointer to member [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weath
erfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(359): error C2109: sub
script requires array or pointer type [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\buil
d\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(359): error C2227: lef
t of '->name' must point to class/struct/union/generic type [C:\Users\Rick\Documents\GitHub\o-plugi
n\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(360): error C2597: ill
egal reference to non-static member 'WeatherFax::m_BuiltinCoords' [C:\Users\Rick\Documents\GitHub\o
-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(360): error C3867: 'We
atherFax::m_BuiltinCoords': function call missing argument list; use '&WeatherFax::m_BuiltinCoords'
 to create a pointer to member [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weath
erfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(360): error C2109: sub
script requires array or pointer type [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\buil
d\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(362): error C2228: lef
t of '.m_CaptureSettings' must have class/struct/union [C:\Users\Rick\Documents\GitHub\o-plugin\s-w
eatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(369): error C2597: ill
egal reference to non-static member 'WeatherFax::m_UserCoords' [C:\Users\Rick\Documents\GitHub\o-pl
ugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(369): error C2440: 'in
itializing' : cannot convert from 'WeatherFaxImageCoordinateList WeatherFax::* ' to 'WeatherFaxImag
eCoordinateList *' [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcx
proj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(371): error C2671: 'We
atherFax::OpenWav' : static member functions do not have 'this' pointers [C:\Users\Rick\Documents\G
itHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(371): error C2661: 'We
atherFaxWizard::WeatherFaxWizard' : no overloaded function takes 4 arguments [C:\Users\Rick\Documen
ts\GitHub\o-plugin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(379): error C2227: lef
t of '->Enable' must point to class/struct/union/generic type [C:\Users\Rick\Documents\GitHub\o-plu
gin\s-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\WeatherFax.cpp(386): error C2228: lef
t of '.push_back' must have class/struct/union [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfa
x_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_pi\src\SchedulesDialog.cpp(799): error C2039
: 'contains' : is not a member of 'wxString' [C:\Users\Rick\Documents\GitHub\o-plugin\s-weatherfax_
pi\build\weatherfax_pi.vcxproj]

    0 Warning(s)
    20 Error(s)

seandepagnier, 2017-03-23:
Rick,

Could you please post the cmake output?

rgleason, 2017-04-04:
Sean, I think you are looking for this output log.  What version of OPENCPN.LIB should we be using?

[CMakeOutput .log.txt](https://github.com/seandepagnier/weatherfax_pi/files/893282/CMakeOutput.log.txt)

rgleason, 2017-06-02:
Using opencpn-4-5-316.lib  as opencpn.lib in the build directory and recent git fetch, git pull, it compiled successfully on my laptop.

Now closing this.

closed 2017-06-02 by rgleason