seandepagnier/weatherfax_pi #46: Weatherfax fails to build on ARMHF Ubuntu 14.04 (Tegra TK1)

sbfreddie, 2017-03-09:
Sean:
My script was building all the plugins and weatherfax build stopped with this error:

/usr/bin/c++   -DBUILTIN_RTLAIS -DOCPN_USE_PORTAUDIO -DTIXML_USE_STL -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D__WXGTK__ -DocpnUSE_GL -Dweatherfax_pi_EXPORTS -pthread -fPIC -I/home/freddie/GitOpenCPN/Seans_Plugins/weatherfax_pi/include -I/home/freddie/GitOpenCPN/Seans_Plugins/weatherfax_pi/src -isystem /usr/lib/arm-linux-gnueabihf/wx/include/gtk2-unicode-3.0 -isystem /usr/include/wx-3.0 -I/home/freddie/GitOpenCPN/Seans_Plugins/weatherfax_pi/./src/libaudiofile    -Wall -Wno-unused-result -g -fexceptions -o CMakeFiles/weatherfax_pi.dir/src/WeatherFax.cpp.o -c /home/freddie/GitOpenCPN/Seans_Plugins/weatherfax_pi/src/WeatherFax.cpp
/home/freddie/GitOpenCPN/Seans_Plugins/weatherfax_pi/src/WeatherFax.cpp:349:19: error: prototype for ‘WeatherFaxWizard* WeatherFax::OpenWav(wxString, AFframecount, wxString, wxString, wxString)’ does not match any in class ‘WeatherFax’
 WeatherFaxWizard *WeatherFax::OpenWav(wxString filename, AFframecount offset, wxString station, wxString area, wxString contents)
                   ^
In file included from /home/freddie/GitOpenCPN/Seans_Plugins/weatherfax_pi/src/WeatherFax.cpp:33:0:
/home/freddie/GitOpenCPN/Seans_Plugins/weatherfax_pi/src/WeatherFax.h:94:23: error: candidate is: WeatherFaxWizard* WeatherFax::OpenWav(wxString, long int, wxString, wxString, wxString)
     WeatherFaxWizard *OpenWav(wxString filename, long offset=0, wxString station=_T(""), wxString area=_T(""), wxString contents=_T(""));
                       ^
make[2]: *** [CMakeFiles/weatherfax_pi.dir/src/WeatherFax.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory `/home/freddie/GitOpenCPN/Seans_Plugins/weatherfax_pi/build'
make[1]: *** [CMakeFiles/weatherfax_pi.dir/all] Error 2
make[1]: Leaving directory `/home/freddie/GitOpenCPN/Seans_Plugins/weatherfax_pi/build'
make: *** [all] Error 2
Installing Weatherfax Failed
********************************************************************************************

Do you have any suggestions for a solution?

Thanks,
Ed

Ulmsp, 2017-03-29:
Sean:
the same Error on RASPBIAN JESSIE LITE
Is there anny solution?

regards Michael

rgleason, 2017-09-07:
This possibly may be the problem, fixed Windows.

https://github.com/seandepagnier/weatherfax_pi/issues/51#issuecomment-327862331

rgleason, 2017-09-15:
Sean has made some adjustments for ARMHF
March 30 
https://github.com/seandepagnier/weatherfax_pi/commit/4f61d187f1495023aec9f29edaa47c80899cf71c
after Domig  PR 
https://github.com/seandepagnier/weatherfax_pi/commit/85e995a771c6aab0c782861105c2199de2be9a87

seandepagnier, 2018-02-08:
I build on ARMHF... so it's fixed?

closed 2018-02-14 by seandepagnier