seandepagnier/weatherfax_pi #47: German Weather Service Charts

free-x, 2017-03-11:
Hi,

is it possible to add the Meteocharts from German Weather Service (DWD) to weatherfax?

http://www.dwd.de/DE/leistungen/hobbymet_wk_europa/hobbyeuropakarten.html?nn=540098

Regards
free-x

LeaseOnLife, 2017-03-11:
Good find!

Rick G (rgleason) has been maintaining the download list including map calibrations , I hope he jumps on it.

@rgleason , in case you don't want to decifer German, below are the direct links for the more common surface analysis and forecasts:

- Surface analysis: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/bwk_bodendruck_na_ana.png 
- 24h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_024.png
- 36h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_036.png
- 48h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_048.png
- 60h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_060.png
- 84h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_084.png
-108h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_108.png

There are more, maybe the 500mbar are worth adding too?

Dirk

seandepagnier, 2017-03-11:
If the url is fixed they can simply be added to the xml with the correct calibration.

rgleason, 2017-03-18:
Yes Sean has a good. Saves me work grin.
Is this using fixed urls?
If yes you can add the new url to the weatherfaxinternetretieval.xml in the same form as others.  No maybe i had better do it. Send me all the url links with english description of what the fax is please, as a list, in the order you want.
Then I will add it to the file and I may send it back for you guys to calibrate with the wizard. Thats work. Ok? 500mb would be good.

rgleason, 2017-03-18:
Dirk, just add 500mb to here and anything else you want

rgleason, 2017-03-18:
Does surf analysis have time period?

rgleason, 2017-03-18:
Any. Chance we can get this working in windows sean? See the issues. Thanks.

LeaseOnLife, 2017-03-18:
Rick, I am happy with the list above. I have not found the surf-analysis, link?

free-x, 2017-03-18:
Hi,

yes, all charts have a fix URLs
additionally 500hPa Links:

- Analyse http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_500ht_na_ana.png
- +24h http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_500ht_na_024.png
- +48h http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_500ht_na_048.png
- +72h http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_500ht_na_072.png

Regards
free-x

rgleason, 2017-03-19:
I'll put this together but it won't be useful until this is fixed!  I can't compile it anymore.

https://github.com/seandepagnier/weatherfax_pi/issues/45

rgleason, 2017-06-23:
Dirk,
I can do this now.  
Do you want to add 500mbar would be worthwhile too?

Best
Rick

LeaseOnLife, 2017-06-23:
I checked the above links, they all still work, are fixed web links:
Surface charts:
- Analysis: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/bwk_bodendruck_na_ana.png 
- 24h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_024.png
- 36h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_036.png
- 48h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_048.png
- 60h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_060.png
- 84h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_084.png
- 108h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_108.png

500hPa (=500mbar):
-   Analysis http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_500ht_na_ana.png
-   +24h http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_500ht_na_024.png
-    +48h http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_500ht_na_048.png
-    +72h http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_500ht_na_072.png

rgleason, 2017-06-23:
Dirk, Hi. You are right on this. Thanks.  Please show me the exact titles you'd like. (Take a look at the other url titles in weatherfax for examples, OK?

Also do we call the main source what  "German DWD Meteo"   -- something short is best, even shorter than this perhaps.

rgleason, 2017-06-23:
Had to check when the commit was made in order to find the correct weatherfax files to use
See https://github.com/seandepagnier/weatherfax_pi/commit/926ebd4f8c02ed2f38fb88097728c2565975bad9
Commit was made Feb 19, 2016.
However I have some later CoordinateSets.xml files dated  10/1/2016 and weatherfaxinternetretrival.xml dated 10/21/2016!

I have compared files and found that I made some corrections which are not in the old PR, so I have to use C:\Users\Rick\Documents\1-Weatherfax-Files\New\weatherfaxinternetretrival.xml  from 10/21/2016 and similar for coordinatesets.xml.

rgleason, 2017-07-30:
I have uploaded these to 
http://www.cruisersforum.com/forums/f134/weatherfax-97533-2.html#post2443447

I see I have to add the  500hPa (=500mbar)  
Coming up next...

rgleason, 2017-07-30:
Ok here are the files updated with the 500mb.  Please remove the pdf and unzip and place in the proper weatherfax directories.

[New-07-30-2017-2-GER-500mb.zip.pdf](https://github.com/seandepagnier/weatherfax_pi/files/1185729/New-07-30-2017-2-GER-500mb.zip.pdf)

Can this issue now be closed?   
Lease on Life you or Sean need to close it!  I can't close it.

seandepagnier, 2017-07-30:
please make a pull request!

rgleason, 2017-07-30:
Sean, I'd be happy to do that when I have the entire world corrected!
Hopefully with help from users.
Well maybe before that depending on what happens... 
I don't want to tire you or bdbcat out with successive changes and there are lots of updates still needed.
I just want some help from other users to get this accomplished.
I am trying to show them how it is done. We need 2-3 guys who know how to do this, and we could check and fix all the URL's in jig time.  OK?

By the way, I made a typo error on the German 500mb. Attached files have it corrected.

[New-07-30-2017-2-GER-500mb-rev1.zip.pdf](https://github.com/seandepagnier/weatherfax_pi/files/1185750/New-07-30-2017-2-GER-500mb-rev1.zip.pdf)

rgleason, 2017-07-30:
Can we close this now. This item is done, but we need to check the entire weatherfaxinternetretrieval file for problems.

closed 2017-07-30 by seandepagnier