seandepagnier/weatherfax_pi #51: Preferences in v1.7 seems to bomb OpenCPN

rgleason, 2017-09-04:
Dear Sean, 
I just get fetched version 1.7 which has the merge of weatherfaxinternetretrieval.xml etc. and when I try to access "Preferences" from the File dropdown or from Options> Plugins > Weatherfax_pi  what happens is the spinner starts spinning and eventually Opencpn bombs.

I attach a file with the last log. There's nothing significant that I can see.

I'd like to get Bdbcat to release this new version, but hesitate to do so with preferences not working.
[opencpn.weatherfax-pref.log.txt](https://github.com/seandepagnier/weatherfax_pi/files/1274831/opencpn.weatherfax-pref.log.txt)

rgleason, 2017-09-06:
I've tried compiling very recent git fetch with opencpn.lib from v4.8.0 and "Preferences" still bombs out Opencpn.   However the rest of the plugin seems to work in Windows, we can download weather faxes etc.

rgleason, 2017-09-06:
I am going to try compiling a debug version. and using it in MSVC++ to get debug info.

The warnings after running debug are:

Build succeeded.

"C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\build\ALL_BUILD.vcxproj" (default targe
t) (1) ->
"C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\build\weatherfax_pi.vcxproj" (default t
arget) (3) ->
(ClCompile target) ->
  C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\src\DecoderOptionsDialog.cpp(47): warn
ing C4800: 'long' : forcing value to bool 'true' or 'false' (performance warning) [C:\Users\Rick\Do
cuments\GitHub\o-plugin\sean-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\src\DecoderOptionsDialog.cpp(48): warn
ing C4800: 'long' : forcing value to bool 'true' or 'false' (performance warning) [C:\Users\Rick\Do
cuments\GitHub\o-plugin\sean-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\inttypes.h(126): warning C4005: 'P
RId64' : macro redefinition [C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\build\weath
erfax_pi.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\inttypes.h(132): warning C4005: 'P
RIi64' : macro redefinition [C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\build\weath
erfax_pi.vcxproj]
  c:\users\rick\documents\github\o-plugin\sean-weatherfax_pi\src\weatherfaximage.cpp(158): warning
C4700: uninitialized local variable 'lastphase_pos' used [C:\Users\Rick\Documents\GitHub\o-plugin\s
ean-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\algorithm(1026): warning C4996: 's
td::_Transform1': Function call with parameters that may be unsafe - this call relies on the caller
 to check that the passed values are correct. To disable this warning, use -D_SCL_SECURE_NO_WARNING
S. See documentation on how to use Visual C++ 'Checked Iterators' [C:\Users\Rick\Documents\GitHub\o
-plugin\sean-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\algorithm(1026): warning C4996: 's
td::_Transform1': Function call with parameters that may be unsafe - this call relies on the caller
 to check that the passed values are correct. To disable this warning, use -D_SCL_SECURE_NO_WARNING
S. See documentation on how to use Visual C++ 'Checked Iterators' [C:\Users\Rick\Documents\GitHub\o
-plugin\sean-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\xutility(2132): warning C4996: 'st
d::_Copy_impl': Function call with parameters that may be unsafe - this call relies on the caller t
o check that the passed values are correct. To disable this warning, use -D_SCL_SECURE_NO_WARNINGS.
 See documentation on how to use Visual C++ 'Checked Iterators' [C:\Users\Rick\Documents\GitHub\o-p
lugin\sean-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\algorithm(1026): warning C4996: 's
td::_Transform1': Function call with parameters that may be unsafe - this call relies on the caller
 to check that the passed values are correct. To disable this warning, use -D_SCL_SECURE_NO_WARNING
S. See documentation on how to use Visual C++ 'Checked Iterators' [C:\Users\Rick\Documents\GitHub\o
-plugin\sean-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\src\libaudiofile\aupv.c(112): warning
C4018: '<' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\b
uild\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\src\libaudiofile\aupv.c(118): warning
C4018: '>' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\b
uild\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\src\libaudiofile\aupv.c(132): warning
C4018: '<' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\b
uild\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\src\libaudiofile\aupv.c(138): warning
C4018: '>' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\b
uild\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\src\libaudiofile\aupv.c(152): warning
C4018: '<' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\b
uild\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\src\libaudiofile\aupv.c(158): warning
C4018: '>' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\b
uild\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\src\libaudiofile\aupv.c(187): warning
C4018: '<' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\b
uild\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\src\libaudiofile\aupv.c(193): warning
C4018: '>' : signed/unsigned mismatch [C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\b
uild\weatherfax_pi.vcxproj]
[... 8 lines cut ...]
uild\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\src\libaudiofile\aupv.c(233): warning
[... cut ...]
  C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\src\libaudiofile\extended.c(109): warn
ing C4244: '=' : conversion from 'unsigned long' to 'unsigned char', possible loss of data [C:\User
s\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\src\libaudiofile\extended.c(110): warn
ing C4244: '=' : conversion from 'unsigned long' to 'unsigned char', possible loss of data [C:\User
s\Rick\Documents\GitHub\o-plugin\sean-weatherfax_pi\build\weatherfax_pi.vcxproj]

    27 Warning(s)
    0 Error(s)

seandepagnier, 2017-09-07:
Rick

Can you get a back trace from the debugger?

Does the version before the merge crash?

Sean

rgleason, 2017-09-07:
Sean, 
Mike Rossiter advised the following after looking at the plugin:

> Rick,
> Start with a fresh git.
> In 'weatherfax_pi.cpp' comment out the lines shown below:
> In ShowPreferencesDialog(wxWindow* parent ) ...
> 
>    dialog->m_sDeviceIndex->SetRange(0, FaxDecoder::AudioDeviceCount() - 1);
> 
> //#ifndef BUILTIN_RTLAIS
> //    dialog->m_cbCaptureType->DeletePage(1);
> //#endif
>    
>     dialog->m_cbCaptureType->SetSelection(m_CaptureSettings.type = FaxDecoderCaptureSettings::RTLSDR);
> 
> I could not find a definition for BUILTIN_RTLAIS.
> Mike

Perhaps this has something to do with your integration of the rtldsr and the rtlais program?

rgleason, 2017-09-07:
The fix above, is working.  Now have access to Preferences and it does not bomb Opencpn!
Thanks Mike.
May work for Debian and others.

rgleason, 2017-09-07:
Sean,
Can you fix this please?  I have asked Bdbcat to release this new version 1.7001 and it would be easier for him if these 3 lines were fixed or commented out.

rgleason, 2017-09-09:
If I get a chance, I 'll try to make a pull request, but it is quite simple.

rgleason, 2017-09-10:
Made a Pull Request #53

rgleason, 2017-09-12:
See Pull Request
https://github.com/seandepagnier/weatherfax_pi/pull/53

rgleason, 2017-09-13:
Closing this, merged

rgleason, 2017-09-14:
My PR #54 copied the files to the root directory rather than the src directory.

rgleason, 2017-09-20:
Can be closed when PR57 is merged.
Close #57

rgleason, 2017-10-15:
Fixed

closed 2017-10-15 by rgleason