seandepagnier/weatherfax_pi #52: Weatherfax fails to build on Debian 9 ( Stretch)

free-x, 2017-09-06:
Hi Sean,

I'm trying to build weatherfax on my Raspberry Pi3 with fresh Raspbian Stretch. But compiling breaks with Error
'''
[ 93%] Building CXX object CMakeFiles/weatherfax_pi.dir/src/libaudiofile/modules/Module.cpp.o
/usr/bin/c++   -DOCPN_USE_PORTAUDIO -DTIXML_USE_STL -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D__WXGTK__ -DocpnUSE_GL -Dweatherfax_pi_EXPORTS -I/home/pi/test/weatherfax_pi/include -I/home/pi/test/weatherfax_pi/src -isystem /usr/lib/arm-linux-gnueabihf/wx/include/gtk2-unicode-3.0 -isystem /usr/include/wx-3.0 -I/home/pi/test/weatherfax_pi/./src/libaudiofile  -pthread -fPIC   -Wall -Wno-unused-result -g -fexceptions -o CMakeFiles/weatherfax_pi.dir/src/libaudiofile/modules/Module.cpp.o -c /home/pi/test/weatherfax_pi/src/libaudiofile/modules/Module.cpp
In file included from /home/pi/test/weatherfax_pi/src/libaudiofile/modules/Module.cpp:26:0:
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h: In instantiation of 'const int signConverter<(FormatCode)0>::kMinSignedValue':
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:130:52:   required from 'signConverter<Format>::UnsignedType signConverter<Format>::signedToUnsigned::operator()(signConverter<Format>::SignedType) [with FormatCode Format = (FormatCode)0; signConverter<Format>::UnsignedType = unsigned char; signConverter<Format>::SignedType = signed char]'
/usr/include/c++/6/bits/stl_algo.h:4177:24:   required from '_OIter std::transform(_IIter, _IIter, _OIter, _UnaryOperation) [with _IIter = const signed char*; _OIter = unsigned char*; _UnaryOperation = signConverter<(FormatCode)0>::signedToUnsigned]'
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:104:16:   required from 'void transform(const void*, void*, size_t) [with UnaryFunction = signConverter<(FormatCode)0>::signedToUnsigned; size_t = unsigned int]'
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:176:62:   required from 'static void ConvertSign::convertSignedToUnsigned(const void*, void*, size_t) [with FormatCode Format = (FormatCode)0; size_t = unsigned int]'
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:183:51:   required from here
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:126:40: error: left operand of shift expression '(-1 << 7)' is negative [-fpermissive]
  static const int kMinSignedValue = -1 << kScaleBits;
                                     ~~~^~~~~~~~~~~~~
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h: In instantiation of 'const int signConverter<(FormatCode)1>::kMinSignedValue':
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:130:52:   required from 'signConverter<Format>::UnsignedType signConverter<Format>::signedToUnsigned::operator()(signConverter<Format>::SignedType) [with FormatCode Format = (FormatCode)1; signConverter<Format>::UnsignedType = short unsigned int; signConverter<Format>::SignedType = short int]'
/usr/include/c++/6/bits/stl_algo.h:4177:24:   required from '_OIter std::transform(_IIter, _IIter, _OIter, _UnaryOperation) [with _IIter = const short int*; _OIter = short unsigned int*; _UnaryOperation = signConverter<(FormatCode)1>::signedToUnsigned]'
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:104:16:   required from 'void transform(const void*, void*, size_t) [with UnaryFunction = signConverter<(FormatCode)1>::signedToUnsigned; size_t = unsigned int]'
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:176:62:   required from 'static void ConvertSign::convertSignedToUnsigned(const void*, void*, size_t) [with FormatCode Format = (FormatCode)1; size_t = unsigned int]'
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:186:52:   required from here
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:126:40: error: left operand of shift expression '(-1 << 15)' is negative [-fpermissive]
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h: In instantiation of 'const int signConverter<(FormatCode)2>::kMinSignedValue':
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:130:52:   required from 'signConverter<Format>::UnsignedType signConverter<Format>::signedToUnsigned::operator()(signConverter<Format>::SignedType) [with FormatCode Format = (FormatCode)2; signConverter<Format>::UnsignedType = unsigned int; signConverter<Format>::SignedType = int]'
/usr/include/c++/6/bits/stl_algo.h:4177:24:   required from '_OIter std::transform(_IIter, _IIter, _OIter, _UnaryOperation) [with _IIter = const int*; _OIter = unsigned int*; _UnaryOperation = signConverter<(FormatCode)2>::signedToUnsigned]'
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:104:16:   required from 'void transform(const void*, void*, size_t) [with UnaryFunction = signConverter<(FormatCode)2>::signedToUnsigned; size_t = unsigned int]'
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:176:62:   required from 'static void ConvertSign::convertSignedToUnsigned(const void*, void*, size_t) [with FormatCode Format = (FormatCode)2; size_t = unsigned int]'
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:189:52:   required from here
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:126:40: error: left operand of shift expression '(-1 << 23)' is negative [-fpermissive]
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h: In instantiation of 'const int signConverter<(FormatCode)3>::kMinSignedValue':
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:130:52:   required from 'signConverter<Format>::UnsignedType signConverter<Format>::signedToUnsigned::operator()(signConverter<Format>::SignedType) [with FormatCode Format = (FormatCode)3; signConverter<Format>::UnsignedType = unsigned int; signConverter<Format>::SignedType = int]'
/usr/include/c++/6/bits/stl_algo.h:4177:24:   required from '_OIter std::transform(_IIter, _IIter, _OIter, _UnaryOperation) [with _IIter = const int*; _OIter = unsigned int*; _UnaryOperation = signConverter<(FormatCode)3>::signedToUnsigned]'
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:104:16:   required from 'void transform(const void*, void*, size_t) [with UnaryFunction = signConverter<(FormatCode)3>::signedToUnsigned; size_t = unsigned int]'
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:176:62:   required from 'static void ConvertSign::convertSignedToUnsigned(const void*, void*, size_t) [with FormatCode Format = (FormatCode)3; size_t = unsigned int]'
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:192:52:   required from here
/home/pi/test/weatherfax_pi/src/libaudiofile/modules/SimpleModule.h:126:40: error: left operand of shift expression '(-1 << 31)' is negative [-fpermissive]
CMakeFiles/weatherfax_pi.dir/build.make:1313: recipe for target 'CMakeFiles/weatherfax_pi.dir/src/libaudiofile/modules/Module.cpp.o' failed
make[2]: *** [CMakeFiles/weatherfax_pi.dir/src/libaudiofile/modules/Module.cpp.o] Error 1
'''
Any Ideas how to repair this issue?

OS: Debian 9.1
g++:  6.3.0 20170516 (Raspbian 6.3.0-18+rpi1)

Regards
free-x

rgleason, 2017-09-07:
See , works for windows.
https://github.com/seandepagnier/weatherfax_pi/issues/51#issuecomment-327862331

rgleason, 2017-09-15:
It looks to me like you do not have the dependencies installed. In particular libaudio.
Please review the installation instructions in README carefully and install the binaries and dependencies needed for your OS.
If you learn anything new about getting it installed on debian please post it here to go into the readme.

Also please see this thread
http://www.cruisersforum.com/forums/f134/installing-on-debian-9-1-a-190654.html

And CaptnRon's experience installing Tactics_pi.
http://www.cruisersforum.com/forums/f134/tactics-plugin-166909.html#post2478899

Please write up your experience to corroborate CaptnRon's so that we can have some documentation for installation of plugins on Debian. 

When this has been resolved, you can close it.  I will be reading anyway. If we can help any further please advise.
Thank you.

free-x, 2017-09-17:
Hi Rick,

are you sure? I can't agree your comment
```
-- Found Portaudio: /usr/lib/arm-linux-gnueabihf/libportaudio.so
-- Portaudio Found
```
Regards
free-x

rgleason, 2017-09-18:
I guess I was wrong. I haven't use Linux much. We've been having trouble with the plugin (Windows) because hitting Preferences bombed window. Mike R. determined what fixed it., see his note.

Start with a fresh git.
In 'weatherfax_pi.cpp' comment out the lines shown below:
In ShowPreferencesDialog(wxWindow* parent ) ... Line 344

dialog->m_sDeviceIndex->SetRange(0, FaxDecoder::AudioDeviceCount() - 1); Line 353

//#ifndef BUILTIN_RTLAIS line 355
// dialog->m_cbCaptureType->DeletePage(1); Line 365
//#endif Line 357

I don't know it this also affects  Linux, as Sean works in linux.

You could try my repository which has this corrected.
https://github.com/rgleason/weatherfax_pi
but no promises for your linux problem.

It does have much more current XML files for MET url's, however.

jcollie, 2017-09-27:
I'm having the same problem on Fedora 26, with GCC 7.2.1. I believe that the problem is that having a negative number on the LHS of a left shift is undefined behavior in the C++ standard. A lot of compilers don't check for that though. New enough versions of GCC do. Adding "-fpermissive" to the compiler flags should work around the problem.

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=68979

free-x, 2017-09-28:
@jcollie THX. That's it! Good catch
I have opened PR #62

rgleason, 2018-01-23:
Pavel wrote 4 days ago.  This can not be closed until the upstream fixes are also pulled to the local copy of libaudiofile..   See https://github.com/seandepagnier/weatherfax_pi/pull/62  at the bottom.

nohal, 2018-01-23:
https://github.com/seandepagnier/weatherfax_pi/pull/62 simply is not good enough and way more has to be pulled in from the [libaudiofile upstream](https://github.com/mpruett/audiofile/), what there is now for example does not build in C++11 mode at all.

rgleason, 2018-01-24:
I may not understand the purpose of this, but is it simply to allow use of rtlsdr  for getting weatherfaxes, or is it for some broader purpose?  I think we simply disabled rtlais for windows.

closed 2018-02-14 by seandepagnier