seandepagnier/weatherfax_pi #63: German DWD no longer works because it is forbidden. Now have to comment out in XML

rgleason, 2017-10-14:
https://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/

We had lots of good links to faxes.. its the only source for Europe at this point...now gone.

See https://github.com/seandepagnier/weatherfax_pi/issues/47
Some of Those links still work.

    Surface analysis: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/bwk_bodendruck_na_ana.png
    24h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_024.png
    36h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_036.png
    48h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_048.png
    60h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_060.png
    84h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_084.png
    108h forecast: http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/ico_tkboden_na_108.png

LeaseOnLife, 2017-10-14:
Rick
try replacing http with https
then the links seem to work, haven't checked all of them, just a few though
Dirk

rgleason, 2017-10-15:
Good Dirk.

rgleason, 2017-10-15:
German DWD no longer works because it is forbidden. Now we have to change the http:// to https:  rather than comment out in XML.
But this is going to be a huge pain in the butt.  Can't the plugin check for https://  if it doesn't work, rather than CRASH OPENCPN.

rgleason, 2017-10-18:
the URL now uses https://  but the situation still exists. 
The plugin bombs O if another URL changes to https:///

rgleason, 2018-01-19:
Has anything been done in the plugin about websites which change from http:// to https://?
Current if the URL has changed to a secured website, the WEBSITE would just crash OpenCPN.
This should not happen! 

Secondly, the plugin should check to see if the URL has become secured as https:// 
and complete if it has, or put out a message....so it can be corrected.

nohal, 2018-01-19:
What does "websites which change from http:// to https://" mean exactly?
Exact steps to reproduce a crash?

rgleason, 2018-01-19:
Pavel, 
There are URL's in the  weatherfaxinternetretrieval.xml  which are written http://.....  and when an external MET weatherfax server changes to https://  we find that the plugin will crash Openpcn.  Dirk discovered the cause of the crash was that the MET had changed the server to use https://  and perhaps did not have redirects to https://   ...or perhaps did,  but the plugin was unable to handle it.

nohal, 2018-01-19:
Example of such an URL? Or do I really have to try all of them before I find one to reproduce the crash?

rgleason, 2018-01-19:
Pavel, 
We have had one instance of this so far.

nohal, 2018-01-19:
But on which of the hundreds of the URLs?

rgleason, 2018-01-19:
Quoted From the top of the Issue

> https://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/
> 
> This website would crash Opencpn
> Probably because the XML uses http:// and the website says "Forbidden" and then O Crashes.
> The website changed to https:// and some point. This will be a pain in the butt.
> Can't the PI test for https://? or if it does not work as http:// for some short period, try https://?
> Also O should not BOMB.

rgleason, 2018-01-19:
I could possibly send you a weatherfaxinternetfaxretrieval.xml file that does bomb, by changing back to URL http://    Would that help?

nohal, 2018-01-19:
This is a https URL. What about the "sites that change from http:// to https://"?

rgleason, 2018-01-19:
Yes, this is a site that made that change, and so after crashing Opencpn, we changed the URL's to https

nohal, 2018-01-19:
Rick...
WHAT DO I HAVE TO DO TO MAKE IT CRASH?

rgleason, 2018-01-19:
Perhaps there should be a test if they changed to https?  with a message?  Rather than crashing O?

rgleason, 2018-01-19:
I haven't done this for awhile Pavel, but I can rename the URL in weatherfaxinternetretrieval.xml
to

http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/

Then try it.   And if it crashes I will post here?

rgleason, 2018-01-19:
Changed to

  <Server Name="German Weather(DWD)" Url="http://www.dwd.de/DWD/wetter/wv_spez/hobbymet/wetterkarten/">

And Tried it.

Now it downloads as well as  the https:// URL
So that site is no longer a good example.

rgleason, 2018-01-19:
Should I just close this now?

nohal, 2018-01-19:
Likely, unless you have a reproducible crash scenario.

rgleason, 2018-01-19:
Pavel, more info here. I tried some other weatherfaxes from the same Server using the http://
and it took over a minute, and a lot of waiting, but eventually a wxfax downloaded.
It looks to me like the behavior is possibly better, because it just sits there trying to connect, rather than crashing...

[image]
[image]

rgleason, 2018-01-19:
So I think I'll close it. It can be reopened later if we find another instance.
Thank youu!

closed 2018-01-19 by rgleason