seandepagnier/weatherfax_pi #79: Passage Weather "Regions" display is slow with spinner and screen blinks

rgleason, 2017-12-19:
See this writeup please:
http://www.cruisersforum.com/forums/f134/weatherfax-97533.html#post2539183

Is is programming or the xml file format?
The release version 1.7005 has this problem and so does 1.7006-beta.
https://github.com/rgleason/weatherfax_pi/releases

rgleason, 2017-12-19:
PS: I also notice that memory use creeps up slowly too.  So when it gets above 700mb I usually shut down the PI and O and start over.  Note that I am processing a lot of images however.

stelian42, 2017-12-19:
Fixed in pull request #82

rgleason, 2018-01-19:
After having checked and updated the first half of the world's Weatherfax URL's I have found that the 12-15 second delay, including a screen black out (low memory), makes doing this work a whole lot slower. I am wishing that the read of URL's and Regions was a whole lot faster.

Regarding memory use while doing these repeative tasks, Opencpn no longer crashes after downloading 100's of images, but memory does creep up to 800 mb -1000mb and the plugin starts refusing to show what it has downloaded. It just won't display the images downloaded.  

Then to regain memory that has been leaked out, I have to close both weatherfax and Opencpn. and restart. Then it downloads fine. I think the recent memory work has helped.

rgleason, 2018-01-29:
Pavel has fixed this by changing the wxWidgets call.  It is much much better. Thank you Pavel!

closed 2018-01-29 by rgleason