seandepagnier/weatherfax_pi #83: Weatherfax fails to be compiled with gcc6

dominig, 2018-01-03:
distros are moving to gcc6 or higher what drags be default more security and control features.
Weather fax is the last of the "official" plugins which still does not compile with gcc6.

Full log is visible at
https://build.opensuse.org/build/home:dominig:opencpn-sandbox/openSUSE_Tumbleweed/x86_64/OpenCPN-weatherfax_pi/_log

------ log extract ----
signConverter<Format>::UnsignedType signConverter<Format>::signedToUnsigned::operator()(signConverter<Format>::SignedType) [with FormatCode Format = (FormatCode)3; signConverter<Format>::UnsignedType = unsigned int; signConverter<Format>::SignedType = int]'
[  189s] /usr/include/c++/7/bits/stl_algo.h:4306:24:   required from '_OIter std::transform(_IIter, _IIter, _OIter, _UnaryOperation) [with _IIter = const int*; _OIter = unsigned int*; _UnaryOperation = signConverter<(FormatCode)3>::signedToUnsigned]'
[  189s] /home/abuild/rpmbuild/BUILD/weatherfax_pi-1513723320.2d2cf8e/src/libaudiofile/modules/SimpleModule.h:104:16:   required from 'void transform(const void*, void*, size_t) [with UnaryFunction = signConverter<(FormatCode)3>::signedToUnsigned; size_t = long unsigned int]'
[  189s] /home/abuild/rpmbuild/BUILD/weatherfax_pi-1513723320.2d2cf8e/src/libaudiofile/modules/SimpleModule.h:176:62:   required from 'static void ConvertSign::convertSignedToUnsigned(const void*, void*, size_t) [with FormatCode Format = (FormatCode)3; size_t = long unsigned int]'
[  189s] /home/abuild/rpmbuild/BUILD/weatherfax_pi-1513723320.2d2cf8e/src/libaudiofile/modules/SimpleModule.h:192:52:   required from here
[  189s] /home/abuild/rpmbuild/BUILD/weatherfax_pi-1513723320.2d2cf8e/src/libaudiofile/modules/SimpleModule.h:126:40: error: left operand of shift expression '(-1 << 31)' is negative [-fpermissive]

--------- end extract ------------------

free-x, 2018-01-23:
Hi,
is the same issue as #52 
fixed in PR #62, but not yet merged.

Regards
free-x

rgleason, 2018-01-23:
Pavel wrote 4 days ago. This can not be closed until the upstream fixes are also pulled to the local copy of libaudiofile.. See #62 at the bottom.

nohal, 2018-01-23:
This is a duplicate of #52 
We certainly do not need a third place to discuss this, we need to pull in the current libaudiofile code (And apply https://github.com/seandepagnier/weatherfax_pi/commit/e42680e206648ac68d1908b7e42b8961a5f020a0, https://github.com/seandepagnier/weatherfax_pi/commit/56b4a7a59bda8eb7d51c68b0bc25c1dfc5a5f577, https://github.com/seandepagnier/weatherfax_pi/commit/2209ffe1de7a59418b49c3af269453b7351a4765, https://github.com/seandepagnier/weatherfax_pi/commit/b9a0305fc0fdf6ef28a03b075c56dce6279848a5 and whatever else is needed to it)...

dominig, 2018-01-23:
Duplicate of #52

closed 2018-01-23 by dominig