seandepagnier/weatherfax_pi #94: Appeyor and Travis failing

rgleason, 2018-02-09:
https://github.com/seandepagnier/weatherfax_pi/commits/master

Sean has accepted and set up his account, but some of the commits fail and have errors.
What should we be doing to fix these?

I assume the result is compiled files for Windows, Linux etc.  but I may be wrong about that.

Is there a good start point to read about this process?

Can we point the download link to this?

seandepagnier, 2018-02-09:
appveyor did give me useful build errors for windows, and I fixed in
the latest commit.

Then the next build failed.   I think it should build now.

I think the automated builds are not reliable.  Sometimes they fail
because they can't download from a certain server.

rgleason, 2018-02-09:
I git fetched your repos and git checkout master, then git reset --hard upstream/master
then pushed to my repos.  Then tried to compile.

> "C:\Users..\weatherfax_pi\build\package.vcxproj" (default target) (1) ->
> "C:\Users\..\weatherfax_pi\build\ALL_BUILD.vcxproj" (default target) (3) ->
> "C:\Users\..\weatherfax_pi\build\weatherfax_pi.vcxproj"  (default target) (4) ->(ClCompile target) ->
>  C:\Users\..\weatherfax_pi\src\libaudiofile\File.cpp(90): error C2039: 'open' : is not a member of '`global namespace'' [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
> C:\Users\..\weatherfax_pi\src\libaudiofile\File.cpp(90): error C2660: 'File::open' : function does not take 3 arguments [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
>   C:\Users\..\weatherfax_pi\src\libaudiofile\File.cpp(121): error C2039: 'close' : is not a member of '`global namespace'' [C:\Users\..\weatherfax_pi\build\weatherfax_pi.vcxproj]
>   C:\Users\..\weatherfax_pi\src\libaudiofile\File.cpp(121): error C2660: 'FilePOSIX::close' : function does not take 1 arguments [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
>   C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\File.cpp(128): error C2039: 'read' : is not a member of '`global namespace'' [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
>   C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\File.cpp(128): error C2660: 'FilePOSIX::read' : function does not take 3 arguments [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
>   C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\File.cpp(133): error C2039: 'write' : is not a member of '`global namespace'' [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
>   C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\File.cpp(133): error C2660: 'FilePOSIX::write' : function does not take 3 arguments [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
>   C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\File.cpp(158): error C2039: 'lseek' : is not a member of '`global namespace'' [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
>   C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\File.cpp(158): error C3861: 'lseek': identifier not found [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
>   C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\openclose.cpp(28): fatal error C1083: Cannot open include file: 'unistd.h': No such file or directory [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
>     8 Warning(s)
>     11 Error(s)
>

rgleason, 2018-02-09:
I find this note in Appveyor in one of the earlier commits that completed successfully for Appveyor:

> Found artifact 'build\weatherfax_pi-1.7.001-win32.exe' matching 'build\*.exe' path
> Uploading artifacts...
> [1/1] build\weatherfax_pi-1.7.001-win32.exe (700,879 bytes)...100%
> "GitHub" deployment has been skipped as environment variable has not matched ("appveyor_repo_tag" is "false", should be "true")
> Build success
> 

How do you get it to "deploy"?    
Would that be done by automatic creation of a new "Release" in your Github Repository?

https://github.com/seandepagnier/weatherfax_pi/releases

What does it require? a Version number assigned or something?

rgleason, 2018-02-09:
For Linux it seems to be ok, but not for appeveyor and Windows.
https://travis-ci.org/seandepagnier/weatherfax_pi/requests

seandepagnier, 2018-02-09:
I have no idea about deploy.  Maybe appveyor can create the nsis
installer?  That would be interesting.

In any case, it looks like it is indeed broken on windows...

rgleason, 2018-02-10:
Sean, so how do I fix these?:

```
 \src\libaudiofile\File.cpp(90): error C2039: 'open' : is not a member of 'global namespace
   \src\libaudiofile\File.cpp(128): error C2660: 'FilePOSIX::read' : function does not take 3 arguments 
    \src\libaudiofile\File.cpp(121): error C2039: 'close' : is not a member of 'global namespace''
     \src\libaudiofile\File.cpp(121): error C2660: 'FilePOSIX::close' : function does not take 1 arguments
   \src\libaudiofile\File.cpp(133): error C2660: 'FilePOSIX::write' : function does not take 3 arguments
     \src\libaudiofile\File.cpp(158): error C2039: 'lseek' : is not a member of '`global namespace''
   \src\libaudiofile\File.cpp(158): error C3861: 'lseek': identifier not found 
   \src\libaudiofile\openclose.cpp(28): fatal error C1083: Cannot open include file: 'unistd.h': No such file or directory
```
I believe libaudio was working before.

rgleason, 2018-02-10:
Sean added an include for Windows. Now Travis seems to work for windows and make a package, but there are some things that still need to be accomplished to be able to download the ompiled file...

```
CPack: - package: /home/travis/build/seandepagnier/weatherfax_pi/build/weatherfax_pi_1.8-1_amd64.rpm generated.
CPack: Create package using TBZ2
CPack: Install projects
CPack: - Run preinstall target for: weatherfax_pi
CPack: - Install project: weatherfax_pi
CPack: Create package
CPack: - package: /home/travis/build/seandepagnier/weatherfax_pi/build/weatherfax_pi_1.8-1_amd64.tar.bz2 generated.
The command "make -j2 package" exited with 0.
Skipping a deployment with the releases provider because a custom condition was not met
Skipping a deployment with the releases provider because this is not a tagged commit
Done. Your build exited with 0.
```

rgleason, 2018-02-10:
I go back to this commit
Fix Headers  https://github.com/seandepagnier/weatherfax_pi/commit/099c8a7b1febe029a8bb1249ba9c5c651c87b90b

And I cannot find any of these files under my local repository github/weaterfax_pi /
Shouldn't they exist somewhere?

 #include <stdlib.h>
 #include <sys/stat.h>
 #include <sys/stat.h>
 #include <sys/types.h>
 #include <sys/types.h>
+#if (HAVE_UNISTD_H)
 #include <unistd.h>
 #include <unistd.h>
+#endif
 #include <stdio.h>
 #include <stdio.h>

Why?

nohal, 2018-02-10:
Rick...
Could you explain what you think is failing on the, obviously successful, [travis build](https://travis-ci.org/seandepagnier/weatherfax_pi/builds/339693014?utm_source=github_status&utm_medium=notification)?

I think opening issues for CI build failures as a whole is a complete waste of effort, everybody sees them and Sean gets notified about the failure automatically and immediately as it happens. Everybody also sees the errors involved in the failure in the build log in the respective CI system, so posting those also has zero added value.

rgleason, 2018-02-10:
Pavel,  
1. I wrote that title earlier, when both Travis and Appveyor were both failing.
2. Right now Appveyor is failing for both linux and windows.
3. Right now Travis for windows does not complete and result in a downloadable exe although it looks like it could.

```
CPack: - package: /home/travis/build/seandepagnier/weatherfax_pi/build/weatherfax_pi_1.8-1_amd64.rpm generated.
CPack: Create package using TBZ2
CPack: Install projects
CPack: - Run preinstall target for: weatherfax_pi
CPack: - Install project: weatherfax_pi
CPack: Create package
CPack: - package: /home/travis/build/seandepagnier/weatherfax_pi/build/weatherfax_pi_1.8-1_amd64.tar.bz2 generated.
The command "make -j2 package" exited with 0.
Skipping a deployment with the releases provider because a custom condition was not met
Skipping a deployment with the releases provider because this is not a tagged commit
Done. Your build exited with 0. 

```

I don't particularly like the fact that we worked very hard to get this plugin updated and working and now it does not compile!   That is just going backwards. 
And so I will write until I understand what is going on and can fix it myself or with the help of others.

rgleason, 2018-02-10:
With Tdan, Did-g & Stelian's help we dragged weatherfax back from the dead and then updated it. You and Sean also made a huge bunch of improvements. Thus my frustration.

So do you mean that Appveyor does only Windows and Travis does only Linux?
I did not know about libaudiofile being old.

closed 2018-02-10 by rgleason

nohal, 2018-02-10:
> So do you mean that Appveyor does only Windows and Travis does only Linux?
Yes, for this plugin, at this moment.
> I did not know about libaudiofile being old.
It is hard to believe looking at stuff like https://github.com/seandepagnier/weatherfax_pi/issues/46 and https://github.com/seandepagnier/weatherfax_pi/issues/52 being commented by you numerous times... But we get to the same stuff as usually. Less is much more in many cases.

rgleason, 2018-02-11:
Pavel,
Now seeing #46 and #52  (and the closing) some of the light bulbs turned on. 

It's going to take me years to learn C++. I wrote Dave to ask him how to start and he gave me some approaches.

I think I am going to back off on all of this now and leave all of you alone.