seandepagnier/weatherfax_pi #96: METs with Dates? - 22 warnings - build succeeds

rgleason, 2018-02-28:
Thought you might want to know. After building 
https://github.com/rgleason/weatherfax_pi/commit/328523f8b8cc7f31f7eb8d52202de47550e49bb5
"add a few changing urls"

Very interesting MET website improvements.   --So we can handle dates?

```

Build succeeded.

"C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build\ALL_BUILD.vcxproj" (de
fault target) (1) ->
"C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj"
 (default target) (3) ->
(ClCompile target) ->
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\DecoderOptionsDialog.cp
p(52): warning C4800: 'long' : forcing value to bool 'true' or 'false' (perform
ance warning) [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build\weatherf
ax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\DecoderOptionsDialog.cp
p(53): warning C4800: 'long' : forcing value to bool 'true' or 'false' (perform
ance warning) [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build\weatherf
ax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\aupv.c(118
): warning C4018: '>' : signed/unsigned mismatch [C:\Users\Frederick\Documents\
GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\aupv.c(138
): warning C4018: '>' : signed/unsigned mismatch [C:\Users\Frederick\Documents\
GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\aupv.c(158
): warning C4018: '>' : signed/unsigned mismatch [C:\Users\Frederick\Documents\
GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\aupv.c(193
): warning C4018: '>' : signed/unsigned mismatch [C:\Users\Frederick\Documents\
GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\aupv.c(213
): warning C4018: '>' : signed/unsigned mismatch [C:\Users\Frederick\Documents\
GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\aupv.c(233
): warning C4018: '>' : signed/unsigned mismatch [C:\Users\Frederick\Documents\
GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\extended.c
(104): warning C4244: '=' : conversion from 'unsigned long' to 'unsigned char',
 possible loss of data [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build
\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\extended.c
(105): warning C4244: '=' : conversion from 'unsigned long' to 'unsigned char',
 possible loss of data [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build
\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\extended.c
(106): warning C4244: '=' : conversion from 'unsigned long' to 'unsigned char',
 possible loss of data [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build
\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\extended.c
(108): warning C4244: '=' : conversion from 'unsigned long' to 'unsigned char',
 possible loss of data [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build
\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\extended.c
(109): warning C4244: '=' : conversion from 'unsigned long' to 'unsigned char',
 possible loss of data [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build
\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\extended.c
(110): warning C4244: '=' : conversion from 'unsigned long' to 'unsigned char',
 possible loss of data [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build
\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\alac\ag_de
c.c(291): warning C4018: '<' : signed/unsigned mismatch [C:\Users\Frederick\Doc
uments\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\alac\ag_de
c.c(307): warning C4146: unary minus operator applied to unsigned type, result
still unsigned [C:\Users\Frederick\Documents\GitHub\weatherfax_pi\build\weather
fax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\alac\ag_de
c.c(325): warning C4018: '<' : signed/unsigned mismatch [C:\Users\Frederick\Doc
uments\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\alac\ag_de
c.c(333): warning C4018: '<=' : signed/unsigned mismatch [C:\Users\Frederick\Do
cuments\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\alac\ag_en
c.c(282): warning C4018: '<' : signed/unsigned mismatch [C:\Users\Frederick\Doc
uments\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\alac\ag_en
c.c(326): warning C4018: '<=' : signed/unsigned mismatch [C:\Users\Frederick\Do
cuments\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\alac\ag_en
c.c(328): warning C4018: '<' : signed/unsigned mismatch [C:\Users\Frederick\Doc
uments\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]
  C:\Users\Frederick\Documents\GitHub\weatherfax_pi\src\libaudiofile\alac\ag_en
c.c(333): warning C4018: '<' : signed/unsigned mismatch [C:\Users\Frederick\Doc
uments\GitHub\weatherfax_pi\build\weatherfax_pi.vcxproj]

    22 Warning(s)
    0 Error(s)

```

seandepagnier, 2018-02-28:
dates, yes

see xml for example

rgleason, 2018-02-28:
Will close now.  Made a comprehensive PR for XML-update with Uwe Aviation, Rick aviaiton polar, Sean strdate examples.

closed 2018-02-28 by rgleason