seandepagnier/weatherfax_pi #106: Wefax schedule update

tchiwam, 2018-08-17:
I just received these 2 wefax from DE, if you want to attach them to your project and/or update the xml description
[image]

[image]

Thank you for the great plugin.

seandepagnier, 2018-08-17:
the hamburg station is already in the schedule list.

rgleason, 2018-08-17:
That is helpful, but it is always easier to have digital text.

tchiwam, 2018-08-18:
These are the schedules coming with the receiver on 
14:25UT 
14:45UT

I noticed these are missing, might be good to map this in a piece of land.

rgleason, 2018-09-14:
Please explain this more completely. I can make the change if necessary, but I need complete information to so. What exactly is needed.

rgleason, 2019-04-12:
This schedule needs to be converted to a text file first and then attached to this Issue so it can be used.

rgleason, 2022-05-22:
I will compare these png's to the current NOAA world wide schedule and make adjustments. This can be closed now.  Thank you.  OP should close.

closed 2022-06-02 by tchiwam