seandepagnier/weatherfax_pi #107: Georeferencing Rotated Images - X is now larger than 1024

C-Rothnie, 2018-09-16:
Hi
I love the Weatherfax plugin for OpenCPN. Combined with a shortwave receiver or internet connection, you can get lots of information very simply. For me it promises to be a better solution than using satellite internet connections for weather forecasts. I really appreciate all the functionality built into the plug-in.

Unfortunately I am struggling with Audio captured images that need to be rotated. The Thai weather charts are transmitted rotated 90 degrees clockwise. There is the option to rotate the image CCW - which is great. However, when geo-referencing the image, you must choose X coordinates (originally Y coordinates) less than the original "Image Width" setting, otherwise the plug-in bombs and you lose your acquired data.

Similarly, when changing the phasing of the acquired data, it is best to do it on non-rotated data and then rotate - otherwise the plug-in can bomb as well.

Finally, the phase and skew settings for different transmitters tends to remain constant, but you need to adjust them every time when using the plug-in because there is no way to store the settings (that I can see). If the phase and skew settings could be saved (or perhaps default to the last used settings), then that would be great too!

Thanks to all contributors for the great plug-in. I like how it is easy enough for sailors to use, but still has a lot of functionality and options to customise.

I attach an acquired image from this morning which I struggled to geo-reference.
[image]
   

Colin Rothnie
Phuket Thailand

rgleason, 2019-04-03:
Is this still a problem?  If not, please close this.  Sean, perhaps you can close this?

rgleason, 2019-04-12:
Now response. Time to close this Sean. I can't.

seandepagnier, 2019-07-18:
possibly still a bug

rgleason, 2022-05-22:
Can this be fixed?