seandepagnier/weatherfax_pi #108: A clean git clone doesn't compile

did-g, 2018-10-13:
Hi,
wxWTranslateCatalog.h isn't generated by cmake anymore. 
 
git show -p 377538cbd40fbd52f40a9d22c2105586c79a56bb cmake/PluginConfigure.cmake 
...
-configure_file(cmake/version.h.in ${PROJECT_SOURCE_DIR}/src/version.h)
-configure_file(cmake/wxWTranslateCatalog.h.in ${PROJECT_SOURCE_DIR}/src/wxWTranslateCatalog.h)
-SET(PACKAGE_VERSION "${PLUGIN_VERSION_MAJOR}.${PLUGIN_VERSION_MINOR}" )
..

dominig, 2018-11-23:
I see the same issue on Linux OpenSUSE and I believe that it's coming from the sale root cause as for weather_routing
see:
https://github.com/seandepagnier/weather_routing_pi/issues/236

rgleason, 2019-01-30:
It seems to be compiling now for Windows, for Opencpn 4.2.0-4.8.8 using wxWidgets 3.0.2
including the new updates for Android, however the menus are working differently and are not as easy to use. Also when opencpn is closed after using this plugin, opencpn seems to crash.

rgleason, 2019-02-19:
https://github.com/seandepagnier/weatherfax_pi/pull/113

rgleason, 2019-04-07:
can't this be closed now? It is months ago and it compiles.
Sean?   OP -did-g?

rgleason, 2019-04-12:
Time to close this!! 
Sean, please close it. I can't.

dominig, 2019-04-19:
Yes it can be close but Dig is the man who can do it.
It's all good.
Thanks.
Dominig
@did-g Didier,

please could you close that bug which is fixed.

closed 2019-04-19 by did-g