seandepagnier/weatherfax_pi #119: Bug - Can no longer switch between can no longer switch between files that are opened in weatherfax

rgleason, 2019-03-27:
There is one problem I discovered (I wrote this to you {Rick} earlier), that you can no longer switch between the files you opened in weatherfax I believe this was from version “I” onwards. As I believe, that this is a real “downgrade”, I think we – I mean Sean – will have to have a look into this. What would be the best way, to get this started?

rgleason, 2019-03-27:
Uwe reports:

So having said this, I use – very heavily – OpenCPN (as this is THE tool on board for me as well and for most of the ww sailing sailors). During these calls I use my prepared “briefing package”, preloaded on OpenCPN and here “WeatherFax”!!!” as you can see on the attached screenshot (besides the GRIBs and some text info you get from meteo-stations).

 

You see in the Weatherfax-window all the charts I am using for the Med and the North Atlantic. Usually I could simply display one of th echarts, by clicking on the entry e.g. “German Weather DWD Surface Analysis” – then you want to see the 24hr forecast – formerly I could simply click on the entry (German Weather DWD – European – 24hr Forecast entry) and it was displayed. This doesn NOT work any longer, you can ONL click on the last entry and here nothing happens (as this chart is already displayed) and when you double click, you open the calibration menus ….
[image]

Due to this I have to switch back to OpenCPN 4.8.8 and use the Weatherfax version, which still works….

stelian42, 2019-03-27:
I confirm there is a problem with clicking on the retrieved fax list.

However, moving through the list using the arrow keys on the keyboard works, and the faxes are displayed just fine. Even transparency works (use shift + arrow to select several).

rgleason, 2019-03-30:
Yes, Stelian, good point, there are workarounds.  Is it possible to get the normal cursor working properly?

rgleason, 2019-04-01:
PS: I reverted this version back to before android changes, see #121  which fixes the v4.8.8 so the cursor works properly and the decoder is sizeable. It does not fix ov50.

stelian42, 2019-04-01:
The following patch (to master) fixes this problem.

Note however this is not the correct way to solve this problem because WeatherFaxUI.cpp is automatically generated by wxFormBuilder.
[wf-patch.txt](https://github.com/seandepagnier/weatherfax_pi/files/3030160/wf-patch.txt)

rgleason, 2019-04-01:
Stelian my branch ov42-fix does compile for ov50 with appveyor but it does not open in ov50. I must be missing several important fixes to wxwidgets for it to run.
Do you have any idea what they might be and how to cherry pick them into a ne branch from this branch? I've got git down pretty much now but this would be new to me.

Or should i just try to fix my ov50 branxh with your file?

stelian42, 2019-04-02:
I'm using WF from git master with OpenCPN from git master too on Linux and it works (well it needs the patch to fix the mouse clicks).

So the problems you have with WF master on Windows must be compile related or Windows related. Is there a bug report specific to this issue ? Are there any obvious errors in log files ?

rgleason, 2019-04-02:
Master does not work for windows. The andriod and mac changes screw things up. The the windows ov50 version is in my ov50 branch. It has both the file selection problem and the wf decoder wizard frozen menu size problem.

rgleason, 2019-04-03:
This bug has been fixed with Stelian's diff file.
https://github.com/seandepagnier/weatherfax_pi/pull/118

closed 2019-04-03 by rgleason