seandepagnier/weatherfax_pi #137: OSX - It won't show in OCPN 5 ->Options -  Plugins

drstvnm, 2020-02-01:
I got the package build on macOS mojave 10.14. 
Departed from the guide at https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_mac_osx in two significant ways that caused compiling failure.
1) installed the wxWidgets 3.1.2. without the -without-subdirs option
2) Manually placed librtlsdr.0.dylib , libportaudio.2.dylib, libusb-1.0.dylib in the build directory.

Package did build and could install in the proper location in OCPN5, but it does not show in the options, list.  
As a test of the development environment i tried the watchdog_pi which compiled and installed fine. Ran out of ideas. I am including the build directory with the package inside....
[build.zip](https://github.com/seandepagnier/weatherfax_pi/files/4142249/build.zip)

rgleason, 2022-05-22:
I believe MacOS is working and this is an old Issue. Can it please be closed, or at least check the current version in the Beta Catalog and report here.  Otherwise the OP should close.

rgleason, 2022-05-23:
Sean, we haven't had a response from @drstvnm  I have confirmation it is working on MacOS in the testing spreadsheet from @Aldebarian https://docs.google.com/spreadsheets/d/19JvjTHTXqzE-atlgQ_SPdqTaAxfCUAiQ5_1vXa1ENb0/edit#gid=0

Sean, please close this, I can't. Thanks.

drstvnm, 2022-05-23:
Rick thanks for taking interest in this issue. I am happy to close it if you could point me to a link with a working version of WF for OCPN 5.

rgleason, 2022-05-23:
The current version is Opencpn 5.6.2  and MacOS works on that.  Sorry we don't go backwards here. The weatherfax_pi v 1.9.32.1 is in the Beta Catalog.  For instructions how to access the Beta Catalog see the Instructions Tab in the Volunteer Testing Spreadsheet (see the link above).   

Test, then report and close this please.  Or Sean?

drstvnm, 2022-05-23:
Tested WeatherFax 1.9.32.1 from the Beta Catalog on Opencpn 5.6.2 in a MacOS Monterey 12.4 environment.
It is a P.

closed 2022-05-23 by drstvnm

drstvnm, 2022-05-23:
> Thank you

Thank you too. 
I am glad I could help a tiny bit.
Stefano