seandepagnier/weatherfax_pi #138: RTL-SDR (feature request, no issue)

Jamos1988, 2020-03-24:
Hi Sean,

I've visited the cruisersforum where the weatherfax is discussed, but couldn't find if this question has been asked yet. Sorry that I am posting this within the issues, let me know if I need to post it somewhere else.

I would really like to be able to control the RTL-SDR dongle from the weatherfax plug-in.
I don't have a radio and not able to use the soundcard, but I do have an extra SDR dongle.
I don't want to use separate software to tune manually to the frequencies, press record, convert with another separate program.

This plug-in and how it works is just way to nice. I realize this request is not that simple, as we would need to let the weatherfax plug-in tune to the frequencies.

I do have some information on how I did this with python scripts, for receiving NOAA and METEOR sattelite images. So I think it is in fact feasible or do-able.

More information about how I did it can be found here: https://www.instructables.com/id/Raspberry-Pi-NOAA-Weather-Satellite-Receiver/ 

Hope you are as enthousiastic as me. :)

Kind regards,

Jamos

seandepagnier, 2020-03-24:
it's already implemented but not very well tested for HF weather fax but you need an upconverter or a way to receive these frequencies.

For other weather faxes like wefax I did not try or implement anything.

Jamos1988, 2020-03-24:
Thats good to hear,  is there a way for me to test it? Some instruction?
And, by upconverter do you mean a piece of hardware? Regards, Jamos

Jamos1988, 2020-03-24:
In opencpn settings I see:

```
vizGENNAV=0
[Settings/WeatherFax]
Path=
ExportPath=
[Settings/WeatherFax/Schedules]
LoadAtStart=1
[Settings/WeatherFax/Capture]
type=1
[Settings/WeatherFax/Capture/audio]
deviceindex=0
samplerate=8000
[Settings/WeatherFax/Capture/rtlsdr]
deviceindex=10
errorppm=117
```
Does this mean I don't have to do anything in the settings and it just finds the rtl-sdr dongle?
I also checked the /home/pi/.opencpn/plugins/weatherfax folder but this is empty.
I can't seem to figure out how I could configure it. Hope you can help again.

seandepagnier, 2020-03-24:
the upconverter is a separate hardware like hamitup which goes before the antenna input.   Otherwise it's possible to use direct mode to get ssb.

Maybe the rtl is not detected?   Do you have rtlsdr working for ais?

I implemented rtlsdr support 3 years go but did not do a lot of testing.  The intention is to schedules which faxes you want and the plugin automatically tunes and captures at the right time, and when complete the image automatically overlays on the chart.   I did not get much feedback and because the rtlsdr has poor reception for hf weather fax I did not spend much more time.

Jamos1988, 2020-03-25:
Ah, I understand. I have two dongles from NooElec. One is being used by AIS indeed and managed from openplotter. The second is free and I tried to give it an extra parameter -d 1 in the preferences of weatherfax. The weatherfax says it cannot claim the RTL_FM after I set the parameter, but I know for sure the device ID = 1. Maybe I need to set the parameter in a different way, but I also tried it in the terminal as follows, so I know it's good:

```
pi@openplotter:~ $ rtl_test -d 1
Found 2 device(s):
  0:  NooElec, NESDR Nano 3, SN: 1112216165
  1:  NooElec, NESDR Nano 3, SN: 2782081715

Using device 1: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
```

So my guess is that the parameter works in terminal, but not in the weatherfax settings.
I do think the feature is great and I am willing to test it thoroughly helping to get the feature more stable if you wish. I'll also order an upconverter if needed.

Jamos1988, 2020-03-25:
My bad, I've set the parameter in rtl-sdr plugin and not weatherfax preferences. So, I guess I need to have an option in weatherfax to use the second dongle with -d 1, or in openplotter somehow setting sdr-ais to -d 1.

You know how to do this?

Jamos1988, 2020-03-31:
Hi Sean, was wondering if you have the time for this. 

-Do you know where the "failed to configure capture" comes from? The RTL dongles are not in use, or claimed by any other process.
-In which file can I configure the -d 1 parameter, so my second dongle is being used?

I am running the opencpn plugin on a raspberry pi.