seandepagnier/weatherfax_pi #147: xcessive runtime linux deps: rtlsdr

rgleason, 2021-04-22:
Sean, 
We need to update weatherfax data files as they are getting out of date. I have two guys that have offered to help.
However in updating the plugin to the new data files and the current FE2 I think we need to address this issue somehow.:
https://github.com/rgleason/weatherfax_pi/issues/48

Do you have any suggestions or workarounds?
There seems to be some problem with  ".. causes problems with the new plugin installer since users manually must install the Debian package for plugin to work."

seandepagnier, 2021-04-23:
Not sure about the issue, but the files, is it just a matter of
re-calibrating the images?

rgleason, 2021-04-23:
Actually there are new urls needed and re-calibration of images and I expect different sets of files will be needed.  Needs to be done regularly I think.