seandepagnier/weatherfax_pi #150: WeatherFaxInternetRetrieval.xml not found

mgrouch, 2021-09-03:
Failed to Download: https://raw.githubusercontent.com/seandepagnier/weatherfax_pi/master/data/WeatherFaxInternetRetrieval.xml
Verify there is a working internet connection.
If the url is incorrect please edit the xml and/or post a bug report.

Thanks

ozolli, 2022-04-25:
This file doesn't exist anymore as it has been split in 12 different files. 
The "Update data sources" function has not been updated to reflect those changes...

However, what is the best way to submit updated data sources ? I have updated the French Polynesia one (sources and geo-referencing).

rgleason, 2022-04-25:
Sean is going to need to accept the very old and very long PR from my repository in order to get those separate files unless he wants me to change the path to be to my repository.  (I wish he would accept the PR)

Ozolli. You can make a PR to my repository for weatherfax with the changes.  Or if you would like you can upload a file and I will get around to making the change.  https://github.com/rgleason/weatherfax_pi

rgleason, 2022-05-22:
This is working now, as it is currently pointed to rgleason/weatherfax_pi/data in the code, however I note that Sean has accepted a very very big PR from my repository, so perhaps we can change back to his repository. 

Since this specific problem is now fixed, the OP should close.

rgleason, 2022-05-22:
@mgrouch  It works now. Time to close.
@seandepagnier  Do you want me to point it back to your repo?

closed 2022-05-23 by mgrouch