seandepagnier/weatherfax_pi #160: Two Top Menus Missing from Weather Fax on MacOS

fourfunfourfun, 2025-03-21:
My Weather Fax window is missing the "Retrieve" and "Help" menu drop downs when on MacOS. 

I have uninstalled and reinstalled the plugin and OpenCPN.  I have installed on two different Mac (M4 Mac Mini and M2 MacBook Air), each with same os and OpenCPN versions and the problem.

WeatherFax 1.10.17
OpenCPN 5.10.2 (Universal)
MacOS Sequoia 15.3.2

rgleason, 2025-03-22:
@fourfunfourfun Can you please provide screenshots of this problem that will show what your problem is?

I have had other MacOS users say it works.

fourfunfourfun, 2025-03-22:
[image]

fourfunfourfun, 2025-03-22:
Also, here is the console information from the last run that has two failed to load messages:

9:02:50.928 MESSAGE base_platform.cpp:273 PlugInManager: Using data dirs from: /Applications/OpenCPN.app/Contents/SharedSupport/plugins;/Users/charlesjordan/Library/Application Support/OpenCPN/Contents/SharedSupport/plugins;/Applications/OpenCPN.app/Contents/PlugIns
09:02:50.928 MESSAGE base_platform.cpp:285 PlugInManager: using data dir: /Users/charlesjordan/Library/Application Support/OpenCPN/Contents/SharedSupport/plugins/weatherfax_pi
09:02:50.929 MESSAGE WeatherFax.cpp:201 Weather Fax : 
09:02:50.938 MESSAGE base_platform.cpp:273 PlugInManager: Using data dirs from: /Applications/OpenCPN.app/Contents/SharedSupport/plugins;/Users/charlesjordan/Library/Application Support/OpenCPN/Contents/SharedSupport/plugins;/Applications/OpenCPN.app/Contents/PlugIns
09:02:50.938 MESSAGE base_platform.cpp:285 PlugInManager: using data dir: /Users/charlesjordan/Library/Application Support/OpenCPN/Contents/SharedSupport/plugins/weatherfax_pi
**09:02:50.938 MESSAGE WeatherFax.cpp:201 Weather Fax : Failed to load coordinate sets
09:02:50.938 MESSAGE WeatherFax.cpp:305 Weather Fax : Failed to load coordinate sets**
09:02:54.529 MESSAGE ocpn_frame.cpp:5476 Got quitflag from SIGNAL
09:02:54.535 MESSAGE plugin_loader.cpp:787 PluginLoader: Deactivating PlugIn: /Users/charlesjordan/Library/Application Support/OpenCPN/Contents/PlugIns/libweatherfax_pi.dylib
09:02:54.559 MESSAGE ocpn_frame.cpp:1691 opencpn::MyFrame exiting cleanly.
09:02:54.562 MESSAGE glChartCanvas.cpp:562 BuildFBO 3
09:02:54.563 MESSAGE glChartCanvas.cpp:562 BuildFBO 3
09:02:54.579 MESSAGE chartdb.cpp:287 Chart cache PlugIn purge
09:02:54.581 MESSAGE ocpn_frame.cpp:1870 gFrame destroyed.
09:02:54.591 MESSAGE ocpn_app.cpp:1948 opencpn::MyApp starting exit.
09:02:54.591 MESSAGE ocpn_app.cpp:1988 LOGBOOK:  2025-03-22 13:02:54 UTC OFF: Lat   33.35800 Lon  -79.28200
09:02:54.591 MESSAGE TCDataSource.cpp:41 UnLoading Tide/Current data source: /Applications/OpenCPN.app/Contents/SharedSupport/tcdata/harmonics-dwf-20210110-free.tcd
09:02:54.592 MESSAGE TCDataSource.cpp:41 UnLoading Tide/Current data source: /Applications/OpenCPN.app/Contents/SharedSupport/tcdata/HARMONICS_NO_US.IDX
09:02:54.593 MESSAGE ocpn_app.cpp:2016 opencpn::MyApp exiting cleanly...

rgleason, 2025-03-22:
Now I recall the issue.

All mac users know this. The top menu is normally found above that, and not in the application window.

Do ask me why but that is how it works.

fourfunfourfun, 2025-03-22:
The menus were in the top menu just like you said. Thank you!

closed 2025-03-22 by fourfunfourfun